chipsalliance/rocket-chip #1099: Question regarding flushing D-Cache in multi-master system

adich0, 2017-11-09:
Hello,

I'm generating a configuration of the Rocket with AHB interfaces and dropping it in a system with another master and an SRAM. I'm performing a write to an array in software (believing that it now lives in SRAM), then shortly after configure the second master to read the contents from SRAM. However, it seems that the contents are getting stuck in the D-Cache, and I'm having trouble figuring out the right steps to flush the cache. 

I believe the solution involves the fence instruction somehow, but maybe I'm using it incorrectly. Below is a snippet of code of something I've tried:

```
   __asm__ __volatile__ ("fence iorw, iorw\n\t");
   for (i = 0; i < 8; i++) {
      u_1_rev[i] = u_1[7-i];
   }
   __asm__ __volatile__ ("fence iorw, iorw\n\t");

```

Any suggestions would be greatly appreciated!

aswaterman, 2017-11-09:
The D$ supports coherence, but it requires the D$ be snooped.   In your case, the other master is not snooping the D$.

As a total hack that I cannot possibly endorse :) you could change this line: https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/rocket/DCache.scala#L721

to ```if (true) {``` then execute the ```fence.i``` instruction to flush the D$.

terpstra, 2017-11-10:
Also, it sounds like you are using AHB for your memory bus, which is cacheable. If you want it not to be cached, you should make the AHB interface attach as a MMIO device.

adich0, 2017-11-13:
@aswaterman  -- So I gave that a try, and fence.i does indeed flush the D$. However, it seems that this somehow prevents external interrupts from triggering an interrupt handler. Still digging into this, but thanks for the suggestion!

@terpstra -- I actually prefer this solution, and following your commit to allow unaligned MMIO address ranges (thanks for that!), I gave this a try; however, as soon as I tried to access another peripheral on the MMIO bus, things blew up. Still investigating this as well, but I suspect this has something to do with the fact that I'm accessing data sections that are stored in the SRAM on the MMIO bus.

closed 2020-02-09 by mwachs5