chipsalliance/rocket-chip #1101: How to configure rocket-chip with separate core and bus clocks?

youchangkim, 2017-11-10:
I just tried to add "new WithAynchronousRocketTiles(8,3) ++" for the Configs.scala in system directory, but nothing happened at the top-level module.
Can anyone tell me how to separate core and bus clock ports at the top?
By the way WithAynchronousRocketTiles should be WithAsynchronousRocketTiles in coreplex/Configs.scala.
There's a typo.

terpstra, 2017-11-10:
The trait only adds a clock crossing. You will need to plumb out the clock+reset signals to the top.
i.e:

tiles.foreach { tile => tile.module.clock := something ... }

youchangkim, 2017-11-11:
@terpstra Thanks for the comment. 
Where should I put that plumbing and I/O definitions for clocks?
Inside ExampleRocketSystem? Or inside ExampleRocketSystemModule?

terpstra, 2017-11-11:
ExampleRocketSystemModule

youchangkim, 2017-11-14:
Thanks @terpstra .
It seems like **tile_inputs** already exists in coreplex/BaseCoreplex.scala for this purpose.
Is there any plan to implement a trait like **HasSeparateCoreClocks**?
And also, is there any way to separate clocks for system bus and peripheral bus?

terpstra, 2017-11-17:
I don't believe there are plans to add HasSeparateCoreClocks to the rocket-chip library directly, as it depends too much on what your SoC should be shaped like. The ExampleRocketSystem is just there to be ... an example!

All you need to do is define a new IO in your own trait and connect tile_inputs to it.

GitZgl, 2020-03-06:
@terpstra As I understand, the periphery device often runs on slow clock, and the core often runs on fast clock. So can you provide an example that separate the system bus clock and peripheral bus clock ?