chipsalliance/rocket-chip #1114: 64-cycle Division in DefaultConfig?

seldridge, 2017-11-17:
I was doing some benchmarking of different instructions and I was noticing that integer divide takes _a lot_ longer than integer multiplication (or floating point multiplication) when using `DefaultConfig`. 

Unless I'm mistakenly understanding the configuration (or am missing where a parameter gets overriden), I think that the default `divUnroll` value of 1 is going to cause default builds of Rocket Chip to have a 64-cycle divider, i.e., [this counter has to count to 64](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/rocket/Multiplier.scala#L128).

Is this intentional, should the default config's `divUnroll` be changed to 8 (like `mulUnroll`), or am I mistaken in what's going on?

terpstra, 2017-11-17:
You understood it correctly. Bumping divUnroll too high will negatively impact your frequency. Div is a slow instruction on all micro-architectures; it's just especially slow on rocket.

seldridge, 2017-11-17:
Thanks @terpstra. 

Current value of `divUnroll` is correctly set at 1. If a user wants something else, then that should be defined in their own configuration (and potentially impacting achievable frequency). 

As all of this is intended, I'll close this.

closed 2017-11-17 by seldridge