chipsalliance/rocket-chip #1116: Jtag debug port does not simulate

ben-k, 2017-11-18:
I added the following line to `system/Configs.scala`:

```
class DefaultJtagConfig extends Config(new WithJtagDTM ++ new DefaultConfig)
```

Then in `vsim` I run:

```
make output/rv64ui-p-addi.out CONFIG=DefaultJtagConfig
```

And it times out.  Is this known not to work or am I missing something?

mwachs5, 2017-11-18:
It won't work like that. The usual simulation flows in rocket-chip need the `fesvr` to be able to connect to the Debug Module Interface, which is consumed by the JtagDTM when you add WIthJtagDTM.

If you add the JTAGDTM, you can use something like the `new_remote_bitbang` branch is trying to get working.

ben-k, 2017-11-18:
Thanks for the pointer - it looks like that PR is being actively worked on so I can wait until it's done to try it out.

IIRC, previous versions of Rocket Chip reconfigured the testharness to match the ports of the DUT so simulation would still work with the Jtag port.  Why was that approach abandoned?

mwachs5, 2017-11-20:
It still does reconfigure the testharness. That's what the PR is using. But, the way of running the test is completely different in that mode, the FESVR doesn't drive the loading and testingt, we use the `riscv-tests/debug/gdbbserver.py` to talk to the running simulation, load programs, and test out some debug communication. The FESVR has never been able to talk to the test harness with the JTAG present.

ben-k, 2017-11-20:
OK, I'll close this in favor of #1088 then.

closed 2017-11-20 by ben-k