chipsalliance/rocket-chip #1120: $RISCV path

chinmayaPesonal, 2017-11-19:
Hi,

I am following installations to get access to rocket-chip generator. My immediate plan is to create a processor and test it on FPGA. I will customize it then.

Please bear with me on basic questions as I am new to using github and also I get lost in Linux often.

One of the initial steps is to set RISCV environment variable.
Here are instructions in the page-
$ export RISCV=/path/to/riscv/toolchain/installation

When I got the rocket-chip areaa from github, it had the riscv-tools directory.
I am using the path to this directory as $RISCV. Is it correct?

With that after updating autoconf on my rhel 6.9 machine, in the riscv-tools directory, I am trying to run ./build.sh. This is the error I get-

_Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
configure: WARNING: libusb-1.x not found, trying legacy libusb-0.1 as a fallback; consider installing libusb-1.x instead
config.status: error: cannot find input file: `Makefile.in'_

Do I need to update packages as mentioned?
Why is the Makefile.in error.

Also I want to quickly get access to verilog, simulator and emulator, what can I skip from these steps and how?

Any help is appreciated.

Regards
Chinmaya

mwachs5, 2017-11-21:
> When I got the rocket-chip areaa from github, it had the riscv-tools directory.
> I am using the path to this directory as $RISCV. Is it correct?

The `$RISCV` variable should point to where you *install* the RISC-V tools. Generally you set it before running the `./build.sh` script inside `riscv-tools`. 
There is more info in the [riscv-tools README](https://github.com/riscv/riscv-tools#quickstart)

> With that after updating autoconf on my rhel 6.9 machine, in the riscv-tools directory, I am trying to run ./build.sh. 

Make sure you have installed all the prerequisites as listed in the README in the same link.

> Also I want to quickly get access to verilog, simulator and emulator, what can I skip from these steps and how?

You really only need the RISCV variable in order to actually compile and run tests. So you can just try setting it to something "fake" and see how far you get.

You can also modify your build.sh to not install everything, you certainly don't need OpenOCD for what you want to do.

chinmayaPesonal, 2017-11-22:
Thanks a lot mwachs5.
After writing this post, I had installed oracle virtualbox on rhel and in the virtualbox i installed ubuntu 1604.
Within ubuntu I had no issues with installing updates and going through all instructions. So the above issue is actually resolved.

I have another small question, for which I will create a new post. That is related to vsim.

closed 2017-11-22 by chinmayaPesonal