chipsalliance/rocket-chip #1121: How to simulate custom C code?

swilkins8, 2017-11-19:
Would someone be able to help me with running the rocket chip using custom C code?

I am working with another individual on creating an RoCC accelerator for calculating the CRC across a block of data. We have the accelerator implemented and added to the configs (we put it in place of the CharacterCount accelerator).

Ideally, we were wanting to use the proxy kernel to take advantage of the printf statements for relatively quick verification that our implementation works. But at this point, we'd settle for a bare-metal output (ideally where the file size doesn't explode to the order of gigabytes).

We wrote and compiled a C program (with riscv's gcc) using the macros in #467, and dumping out the resulting file creates assembly that looks quite reasonable. We also created a simple program that just adds a couple integers together (like 4 lines of C code), in case something was wrong with our accelerator integration. However, trying to run either of them, either via the proxy kernel or bare-metal seems to never finish, at least after 100 million cycles.
Running the riscv-tests completes without issues.

We've tried running these commands:

```
./emulator-freechips.rocketchip.system-CRCAcceleratorDefaultConfig +max-cycles=100000000 +loadmem=./TestFiles/simpleTest.hex none
./emulator-freechips.rocketchip.system-CRCAcceleratorDefaultConfig +max-cycles=100000000 ./TestFiles/simpleTest
./emulator-freechips.rocketchip.system-CRCAcceleratorDefaultConfig +max-cycles=100000000 pk ./TestFiles/simpleTest
```

And they all error out after 100 million cycles, without completion. Is there a step we are missing here?
Our repo is at https://github.com/swilkins8/rocket-chip-RSA_CRC if looking at it would be of any help here.

swilkins8, 2017-12-01:
We have since been able to run compiled C code.

closed 2017-12-01 by swilkins8

hongber, 2018-08-02:
Do you modified the file: " machine/minit.c ", modified the XS (extension) bits of the MSTATUS register?Thank you.