chipsalliance/rocket-chip #1124: Accessing Verilog of the Core

chinmayaPesonal, 2017-11-22:
Hi,

I followed all steps till
$ cd vsim
$ make

Here I could n't go ahead due to not having bought synopsys vcs licence.

My question is, even without vcs, are all generated verilog complete?
Once I have verilog, I would like to implement it on FPGA. What should I do to get verilog without having to care about vcs?

Thanks
Chinmaya

CLappin, 2017-11-22:
Hi,

The code should still be generated in vsim/generated-src folder. VCS is only needed to synthesize the verilog code, this could be done in an alternative program such as Micorsemi Libero SoC or Xilinx ISE.

Thanks, 
Ciaran

seldridge, 2017-11-22:
Inside `vsim`, the default build target is going to run the Chisel compiler to generate CHIRRTL, run the FIRRTL compiler to generate Verilog, and then run VCS to "compile" the Verilog (along with some other files) to an executable. This is equivalent to the `emulator` directory except that VCS is substituted for Verilator. If you want to bail out early and just generate Verilog, the `verilog` target is provided by the Makefile.

Note, if you want a base FPGA project, you may consider looking at either of Berkeley's [`fpga-zynq`](https://github.com/ucb-bar/fpga-zynq) repository (which will eventually be deprecated) or at [`midas-top-release`](https://github.com/ucb-bar/midas-top-release). 

There are a [few more details in the relevant section of the README](https://github.com/freechipsproject/rocket-chip#fpga).

chinmayaPesonal, 2017-11-24:
Thank you @CLappin and @seldridge .
Yes I saw that the verilog is created. I can access the verilog.

I want to implement the core on FPGA to see it's components and think about how it can be used.

While implementing, I found some issues, for which I will check answers first and create another thread if they have not been asked.

Before I close this, Can you please give me some more info on the fpga-zynq dir. Does it use the zynq arm core for anything?

Another Question :  I read that Chisel generates files for emulation, asic implementation and fpga implementation. Why generated rtl is different for asic and fpga and how?

Regards
Chinmaya

sequencer, 2022-04-08:
> My question is, even without vcs, are all generated verilog complete?

Yes.

> Once I have verilog, I would like to implement it on FPGA. What should I do to get verilog without having to care about vcs?

See sifive/fpga-shell, we also need a full documentation flow on how to get the FPGA bitstream.

closed 2022-04-08 by sequencer