chipsalliance/rocket-chip #1125: Verilog build error

CLappin, 2017-11-22:
Hi,

I'm having a similar issue to #1064 , using commit a8d573beeb8a33bb5dac0673ec68b82852a24d63 

java.security.AccessControlException: access denied ("java.io.FilePermission" "/home/user2/rocket-chip/firrtl/src/test/resources/integration" "readlink")
	at java.security.AccessControlContext.checkPermission(AccessControlContext.java:472)
...

.java:622)
	at java.lang.Thread.run(Thread.java:748)
[error] java.security.AccessControlException: access denied ("java.io.FilePermission" "/home/user2/rocket-chip/firrtl/src/test/resources/integration" "readlink")
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? q

Any help would be greatly appreciated.
Thanks,
Ciaran

CLappin, 2017-11-22:
Note: I built the build-rv32ima.sh and was attempting to build the default config.

colinschmidt, 2017-11-22:
Do you have read access to all of the subdirectories of rocket-chip?

CLappin, 2017-11-23:
I do. I took out the latest code from today and it built fine.  May have been a flaw in my installation.

closed 2017-11-27 by hcook