chipsalliance/rocket-chip #1141: AXI4 Interface Resources

amerlo94, 2017-12-04:
Hi all,
I am synthesizing the rocket core for an FPGA design with AXI4 interfaces. I found that three modules are taking lots of board resources, and so I am asking how could I implement them in a more efficient way. The modules are:

AXI4Deinterleaver
TLBroadcast
AXI4UserYanker

For example In the Verilog generated file the AXI4Deinterleaver is mapped to several queues, is there any behavioral model of these queue, like the sram ones, in order to map them to technology specific blocks of the boards?

These modules are configurable? 

Thank you for the support,
Andrea

terpstra, 2017-12-04:
Ok, so these are all currently a bit ASIC-focussed components.

TLBroadcast: by default it has one cache-block-sized Queue per in-flight transaction (default 4). You can can try the 'DefaultBufferlessConfig' option to slightly reduce performance, but decrease area.

The AXI4UserYanker puts down 2 queues per AXI ID. To reduce the cost of these, you can just reduce the number of AXI IDs you use in the design. Just decrease the number of 'idBits' on the port you want to shrink.

The AXI4Deinterleaver is needed because the AXI4 protocol allows slaves to answer requests with interleaved data response on the 'R' channel. For example, two four-beat reads, X and Y might come back like this: XXYXYXYY. If you can guarantee that your AXI4 slaves do not exploit this corner-case, then you can set the 'interleavedId' of the port to Some(0) and omit the AXI4Deinterleaver entirely. Otherwise, decreasing the number of AXI IDs will also reduce the cost of this component.

Finally, I will note that the AXI4Deinterleaver and AXI4UserYanker both need a MultiQueue abstraction. They enqueue and dequeue at most one element per cycle to one of N queues. At the moment this is implemented using N queues. For an ASIC we calculated that even if all the queues shared storage, it would not be worth using an SRAM. For an FPGA, that conclusion is reversed. So, if you want to improve rocket, we could definitely use a MultiQueue that shares storage into one SRAM.

amerlo94, 2017-12-12:
Thank you for the suggestion.

I was able to save some cells with out tips, but I am still facing huge AXI4UserYanker depth queue (64-bit ram with 20bits as addressing).

Is there anything I could do to have them smaller?

terpstra, 2017-12-12:
20 bits of address??? You've configured something wrong somewhere. It
should be like 1 to 8 deep at most and less than 20 bits wide.

amerlo94, 2017-12-13:
Yes, it was my mistake.

The biggest queue that I see its 18 bits wide with 32 elements.
Is there any documentation about the sourceBits parameters in the AXI4 interfaces?

terpstra, 2017-12-16:
Most of rocket code, including the AXI4 support, is self-documenting (a polite way to say not documented at all)!

closed 2017-12-16 by terpstra