chipsalliance/rocket-chip #1147: Fixed delay for first memory access

jm4rtin, 2017-12-07:
When running code that performs an AHB or memory access, the rocket chip seems to delay execution for a fixed amount of time.
[image]

This happens until around a time of 0x1000:
[image]

Do you know what is causing this?

jm4rtin, 2017-12-07:
I'm using a tiny core configuration. If I reduce this size:
`new WithL1DCacheSets(4096)     ++ // 256kb RAM`

by half it takes until the time reads 0x800 instead (which is half the time). The memory should be a scratch but it seems like it is in some way initializing something or otherwise waiting an amount of time proportional to the size of the memory. The memory contents are not changed.

aswaterman, 2017-12-07:
The D$ has a state machine to initialize the tag RAM, and it blocks memory accesses during that procedure.  The time is proportional to the number of lines in the cache.

jm4rtin, 2017-12-07:
Shouldn't there be no tag cache when using a scratchpad instead of a D$? Isn't the tag_array only used in the else case for usingDataScratchpad?

aswaterman, 2017-12-07:
Yeah, that's a performance bug we haven't bothered to fix, because it only affects reset behavior.  But since you've brought it up, I guess we should do something about it.

jm4rtin, 2017-12-07:
I just changed flushDone:
`val flushDone = if(usingDataScratchpad) Bool(true) else (flushCounterNext >> log2Ceil(nSets)) === nWays`

Of course it's probably not optimal and may not even be right (I haven't looked into how to run the tests yet) but for now it was simple enough and my code isn't waiting for a whole lot of nothing. Reading the Chisel in the rocket can be quite challenging.

aswaterman, 2017-12-07:
I think that's correct but not optimal.  It'll be fixed the right way later today.

aswaterman, 2017-12-08:
Resolved by #1148

closed 2017-12-08 by aswaterman