chipsalliance/rocket-chip #1152: A reset in 1-3 cycles does not reset the system

pentin-as, 2017-12-12:
[image]

[image]

[image]

In the picture, the signals req and resp in the frontend.
As you can see, the frontend seems to start executing from the start address, but the core immediately returns the PC value that was before the reset.

A reset of 4 cycles works.

rocket-chip: 5155eb60597581fd9933ad8b8f864b4416124776

aswaterman, 2017-12-12:
This might not be written down anywhere, but we have been following the Xilinx guideline that blocks must be held in reset for 16 cycles.

pentin-as, 2017-12-12:
I do:
 - setting the reset signal to 16 cycles;
 - zeroing of the bss section (total RAM);
 - loading the test;
 - release of the reset;
 - waiting for the appearance of something in the "to_host";
 - repeat with new test.

Tests without virtual memory pass all.
Tests with virtual memory, if I restart the simulation, pass all.
But if I run sequentially (through reset and zeroing bss) do not pass.
For example, if I run first rv64ui-v-addiw, then rv64ui-v-addw, then the rv64ui-v-addw test will display the message:
"Assertion failed: pt [3] [addr / (1UL << 12)] & 0x080"
it's: https://github.com/riscv/riscv-test-env/blob/68cad7baf3ed0a4553fffd14726d24519ee1296a/v/vm.c#L106

It seems to me that something in the rocket in the part of the TLB is not reset.

aswaterman, 2017-12-12:
I'm confused.  You said resetting for 1-3 cycles fails, but 4 works, and 16 doesn't work.  That is a surprising inconsistency.

I know our reset strategy works, because it's in regression testing.

aswaterman, 2017-12-12:
I'll be out of town; @yunsup will help with this.

pentin-as, 2017-12-12:
No. Resetting less than 4 clock cycles, does not reset the rocket at all (analyze the performance of the tests there is no point).
A reset of 16 clock cycles resets the rocket, but does not pass the test sequence with virtual memory.

mwachs5, 2017-12-12:
How are you loading the test? Through fesvr or some other means? I don’t
think the “through fesvr” method works while reset is asserted so not sure
exactly what you are doing.

pentin-as, 2017-12-12:
No. I generated the verilog-code in the vsim folder and tested it. Tests I load in AXI4RAM using verilog function $readmemh. Tests are loaded when the rocket is under reset.
I'm using Cadence CAD.

pentin-as, 2017-12-13:
Hmm. The memcmp function does not compare information.

test dump:
    80002168:	fff54783          	lbu	a5,-1(a0)
    8000216c:	fff5c703          	lbu	a4,-1(a1)
    80002170:	00e79a63          	bne	a5,a4,80002184 <memcmp+0x74>
...
    80002184:	40e7853b          	subw	a0,a5,a4
    80002188:	00008067          	ret

execution:
C0:      10314 [1] pc=[ffffe02168] W[r15=0000000000000000][1] R[r10=0000000000002001] R[r31=ffffffffffe03003] inst=[fff54783] DASM(fff54783)
C0:      10315 [1] pc=[ffffe0216c] W[r14=00000000000000b3][1] R[r11=ffffffffffe02001] R[r31=ffffffffffe03003] inst=[fff5c703] DASM(fff5c703)
...
C0:      10317 [0] pc=[ffffe0216c] W[r 0=ffffffffffe02000][0] R[r11=ffffffffffe02000] R[r31=ffffffffffe03003] inst=[fff5c703] DASM(fff5c703)
C0:      10318 [1] pc=[ffffe02170] W[r 0=0000000000000000][0] R[r15=0000000000000000] R[r14=00000000000000b3] inst=[00e79a63] DASM(00e79a63)
...
C0:      10341 [0] pc=[ffffe02170] W[r 0=0000000000000000][0] R[r15=0000000000000000] R[r14=00000000000000b3] inst=[00e79a63] DASM(00e79a63)
C0:      10342 [1] pc=[ffffe02184] W[r10=ffffffffffffff4d][1] R[r15=0000000000000000] R[r14=00000000000000b3] inst=[40e7853b] DASM(40e7853b)
C0:      10343 [1] pc=[ffffe02188] W[r 0=ffffffffffe0218c][1] R[r 1=ffffffffffe027a0] R[r 0=0000000000000000] inst=[00008067] DASM(00008067)
...
C0:      10346 [0] pc=[ffffe02188] W[r 0=ffffffffffe0218c][0] R[r 1=ffffffffffe027a0] R[r 0=0000000000000000] inst=[00008067] DASM(00008067)

Why can 0 be read from address 0x2001, and 0xb3 from 0xff..fe02001?
Why is there a note from the previous test?
I will continue to analyze.

pentin-as, 2017-12-14:
TLB:
[image]

I will continue to analyze.

pentin-as, 2017-12-14:
I am very sorry. I found error in my testbench.
In addition, thanks for saying that the reset should be 16 cycles.

closed 2017-12-14 by pentin-as