chipsalliance/rocket-chip #1160: FIRRTL Bump breaks code in JTAG Tap

mwachs5, 2017-12-18:
The currently submoduled version of Chisel/FIRRTL breaks the JTAG TAP controller (which is one of the few modules in the codebase written in chisel3._ . It blows away the `irReg` due to the way it interprets something in the JtagTapController and/or DebugTransportModuleJTAG.

In the "pre-bump" Chisel, there is a line in `DebugTransportModuleJTAG` FIRRTL output:
```
JtagTapController.io is invalid
```

That line is not present in the new version. It's not SUPPOSED to have to be present. But it seems to blow things away as a result in the output Verilog. And there is no complaint. Will file a relevant bug on FIRRTL as well. If this line is manually added to the FIRRTL, then things work as expected.

Note that reverting to `Chisel._` works as a fix.

FYI @ducky64

mwachs5, 2017-12-20:
For rocket-chip this issue is fixed by using `Chisel._` instead of `chisel3._`. Since the issue is already opened in FIRRTL, I am going to close this one since it is not actually affecting rocket-chip.

closed 2017-12-20 by mwachs5

mwachs5, 2019-11-21:
@jackkoenig is there reason to believe that this issue was resolved in FIRRTL so that it is no longer a problem when converting the code to chisel3?

jackkoenig, 2019-11-21:
Yes this has been fixed since https://github.com/chipsalliance/rocket-chip/commit/6bd4103c06d728707af8d4a2d7443f59b97f9228, but now with Async Reset we can both convert to `import chisel3._` and use the native async reset support!

commit 07540e3c (2017-12-21): Bump chisel3 and firrtl to get bug fixes

Fixes #1160
PR #1163 (2017-12-21): Bump chisel3 and firrtl to get bug fixes
Fixes #1160
commit 7385c994 (2017-12-21): Bump chisel3 and firrtl to get bug fixes (#1163)

Fixes #1160