chipsalliance/rocket-chip #1164: Project loading failed in building emulator

chingyi071, 2017-12-21:
I am a student building rocket-chip on Ubuntu
I installed riscv-tool and verified with ```spike pk hello```
But when I build emulator through ```make``` in ```emulator```
I encountered the following error message
```
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /root/rocket-chip/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
Using addons: 
[error] java.nio.file.InvalidPathException: Malformed input or input contains unmappable characters: /root/rocket-chip/riscv-tools/riscv-openocd/tcl/target/??1879x??1??.cfg
[error] 	at sun.nio.fs.UnixPath.encode(UnixPath.java:147)
[error] 	at sun.nio.fs.UnixPath.<init>(UnixPath.java:71)
[error] 	at sun.nio.fs.UnixFileSystem.getPath(UnixFileSystem.java:281)
[error] 	at java.io.File.toPath(File.java:2234)
[error] 	at org.eclipse.jgit.util.FileUtils.getFileAttributesPosix(FileUtils.java:688)
[error] 	at org.eclipse.jgit.util.FS_POSIX.getAttributes(FS_POSIX.java:270)
[error] 	at org.eclipse.jgit.treewalk.FileTreeIterator$FileEntry.<init>(FileTreeIterator.java:365)
[error] 	at org.eclipse.jgit.treewalk.FileTreeIterator.entries(FileTreeIterator.java:242)
[error] 	at org.eclipse.jgit.treewalk.FileTreeIterator.<init>(FileTreeIterator.java:227)
[error] 	at org.eclipse.jgit.treewalk.FileTreeIterator.createSubtreeIterator(FileTreeIterator.java:233)
[error] 	at org.eclipse.jgit.treewalk.AbstractTreeIterator.createSubtreeIterator(AbstractTreeIterator.java:572)
[error] 	at org.eclipse.jgit.treewalk.TreeWalk.enterSubtree(TreeWalk.java:1210)
[error] 	at org.eclipse.jgit.treewalk.TreeWalk.next(TreeWalk.java:828)
[error] 	at org.eclipse.jgit.lib.IndexDiff.diff(IndexDiff.java:437)
[error] 	at org.eclipse.jgit.lib.IndexDiff.diff(IndexDiff.java:377)
[error] 	at org.eclipse.jgit.lib.IndexDiff.diff(IndexDiff.java:559)
[error] 	at org.eclipse.jgit.lib.IndexDiff.diff(IndexDiff.java:377)
[error] 	at org.eclipse.jgit.lib.IndexDiff.diff(IndexDiff.java:559)
[error] 	at org.eclipse.jgit.lib.IndexDiff.diff(IndexDiff.java:377)
[error] 	at org.eclipse.jgit.api.StatusCommand.call(StatusCommand.java:148)
[error] 	at com.typesafe.sbt.git.JGit.hasUncommittedChanges(JGit.scala:88)
[error] 	at com.typesafe.sbt.SbtGit$.$anonfun$buildSettings$20(SbtGit.scala:121)
[error] 	at com.typesafe.sbt.SbtGit$.$anonfun$buildSettings$20$adapted(SbtGit.scala:121)
[error] 	at com.typesafe.sbt.git.DefaultReadableGit.withGit(ReadableGit.scala:39)
[error] 	at com.typesafe.sbt.SbtGit$.$anonfun$buildSettings$19(SbtGit.scala:121)
[error] 	at com.typesafe.sbt.SbtGit$.$anonfun$buildSettings$19$adapted(SbtGit.scala:121)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error] 	at sbt.internal.util.EvaluateSettings$MixedNode.evaluate0(INode.scala:214)
[error] 	at sbt.internal.util.EvaluateSettings$INode.evaluate(INode.scala:159)
[error] 	at sbt.internal.util.EvaluateSettings.$anonfun$submitEvaluate$1(INode.scala:82)
[error] 	at sbt.internal.util.EvaluateSettings.sbt$internal$util$EvaluateSettings$$run0(INode.scala:93)
[error] 	at sbt.internal.util.EvaluateSettings$$anon$3.run(INode.scala:89)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] java.nio.file.InvalidPathException: Malformed input or input contains unmappable characters: /root/rocket-chip/riscv-tools/riscv-openocd/tcl/target/??1879x??1??.cfg
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? 

```

mwachs5, 2017-12-21:
There are some Russians characters in riscv-openocd files that break SBT.
But we removed them, are you sure you have updated your riscv-tools to the
correct version (submodule update —recursive)?

chingyi071, 2017-12-21:
@mwachs5 Thanks for your reply. I use the commit ```7ed0445b987bc8b39f2e38ab31847d39e2aceae3``` in my ```riscv-tools```
Is this version okay?

mwachs5, 2017-12-21:
You should always use the one tagged in this repo. Currently. that is `09a1ffe`. So you should use that one.

chingyi071, 2017-12-23:
I found that file with Russian character was removed after the latest tag in repo ```openocd```. So I checkouted to that commit and solved the problem

closed 2017-12-23 by chingyi071