chipsalliance/rocket-chip #1167: Question about using tilelink for the module interface

RaulGoal, 2017-12-23:
Hi, I am a beginner in Chisel and Rocket chip realm. Therefore, there are some questions.

1.TlleLink: A DMA module can leverage AXI interface to process read and write operation with memory in same cycle by read and write channel separately. In similar situation, can DMA  process read and write operation with memory in same cycle by utilizing TileLink uncached interface?

2.Due to being a beginner, may I ask another question: Is there  a simple example for wrapping the module as DMA with Tilelink uncached interface and communicating with memory?

terpstra, 2017-12-23:
1. TileLink does not have independent read and write bus interfaces. However, due to the way bursts work, you can come fairly close to the AXI throughput. Assuming 4-beat bursts, on the TileLink A channel you could have this pattern: RWWWWRWWWWRWWWW.... while on the D channel you have this pattern: RRRRWRRRRWRRRRWRRRR... So you can get 80% of the bandwidth in this case.

2. Are you saying you already have an AXI DMA module or plan to implement a TileLink DMA module?

RaulGoal, 2017-12-23:
Hi, terpstra
About question 2:  I will plan to implement a TileLink(uncached) DMA module

terpstra, 2017-12-23:
You will probably want a control port interface, of which devices/tilelink/Clint.scala can give you an example + a master interface of which the simplest example we have is tilelink/PatternPusher.scala.

RaulGoal, 2017-12-23:
Hi, terpstra
Thanks for your reply.
By the way, are there some tutorials or documents for teaching how to use rocket chip and slightly more detailed interpretation for implementation code as Lazy module and so on?
I have studied Chisel tutorial materials and some information in Chisel wiki. However, there is big gap for me as being beginner to understand rocket chip implementation.
It would be nice for the beginner to understand and utilize rocket chip if there are some helpful tutorials.

terpstra, 2017-12-24:
I agree that more documentation would be helpful.

closed 2017-12-24 by terpstra