chipsalliance/rocket-chip #1168: Failure when building emulator

truhlikfredy, 2017-12-28:
I try to follow the build and I encountered this error when building emulator:

sbt.librarymanagement.ResolveException: unresolved dependency: antlr#antlr;2.7.7:

I have maven 3.0.5-3, java 1.8.0_45 running on Debian 8 64bit. I tried to purge my local maven repository to see if thathas any effect and made no impact. Any pointer what I have done badly?

Kallil12, 2018-01-17:
Hi, did you solve the problem?

AntonKrug, 2018-01-17:
No, but then I tried Debian 9 and Java9 and that was not good as well. Looks like it doesn't like Java 9 as well.

michael-etzkorn, 2022-04-20:
Because of the age of this issue, I'm going to close it. The working group discussed a plan to open a meta issue about supporting different build environments. If, even now, you feel RC maintainers should especially look at this particular build failure, feel free to reopen the issue.

closed 2022-04-20 by michael-etzkorn