chipsalliance/rocket-chip #1169: the result of the output file

zhpzhp, 2017-12-30:
Hi,

The output file in the readme is like:

> C0: 483 [1] pc=[00000002138] W[r 3=000000007fff7fff][1] R[r 1=000000007fffffff] R[r 2=ffffffffffff8000] inst=[002081b3] add s1, ra, s0
> C0: 484 [1] pc=[0000000213c] W[r29=000000007fff8000][1] R[r31=ffffffff80007ffe] R[r31=0000000000000005] inst=[7fff8eb7] lui t3, 0x7fff8
> C0: 485 [0] pc=[00000002140] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00000000] unknown

But the result I got is like:

> C0:          0 [0] pc=[a57f891ff4] W[r 0=0000000000000000][0] R[r 7=0000000000000000] R[r10=87c84523a89d2fbf] inst=[a6a33b47] DASM(a6a33b47)
> C0:          1 [0] pc=[a57f891ff4] W[r 0=0000000000000000][0] R[r 7=0000000000000000] R[r10=87c84523a89d2fbf] inst=[a6a33b47] DASM(a6a33b47)
> C0:          2 [0] pc=[a57f891ff4] W[r 0=0000000000000000][0] R[r 7=0000000000000000] R[r10=87c84523a89d2fbf] inst=[a6a33b47] DASM(a6a33b47)

Is there anything I missed?

aswaterman, 2017-12-31:
Note the "483" in the first line you qouted--that's the cycle number.  The quote in the readme is skipping the first few hundred lines of output.

zhpzhp, 2018-01-02:
I mean the last part in each line, why I only got something like DASM(a6a33b47) but this part in the readme is assembly code?

edcote, 2018-01-09:
@zhpzhp Try running your log through spike-dasm and report back.

closed 2018-04-10 by hcook