chipsalliance/rocket-chip #1172: Question about Rocket chip utilization?

RaulGoal, 2018-01-04:
Hello, I have some questions for my project with Rocket chip.
Description for my project:
1.Architecture: With the diagram shown below, I want to leverage single core with mmio peripherals. A simple adder and SRAM are attached on mmio crossbar. An adder possesses slave interface for configuration from CPU , eg: how many times to process addition, and master interface to fetch data from SRAM as DMA. In addition, the adder is also attached on interrupt bus. SRAM is wrap slave interface only.
2.What I want to do: First, CPU configures adder through mmio. Then, adder fetches data and do calculation. After completing all calculation, adder send an interrupt though interrupt bus to inform  CPU.
[image]

My questions:
1. From my realization, when simulation is executed, the data would load into RAM as shown above with riscv-fesvr. CPU starts operation after finishing load processing. I want to initial  data into SRAM  before starting simulation. Are there any methods as utilizing riscv-fesvr or somethingelse to reach it? If it is possible, how can I achieve?

2. I found an example to attach module on interrupt bus in implementation.
Furthermore, I would simulate the project with C program:
........
int main{
  CPU configures adder;
  adder processes calculation(sends back interrupt after finishing);
  if (CPU gets interrupt signals){
     do something   
  }
}
Are there any apis or methods for checking whether CPU receives interrupt signal in the C program?

terpstra, 2018-01-04:
Re 2: You can either poll the MEIP CSR bit or setup an interrupt handler and enable interrupts (harder). In both cases you will need to write a bit of inline assembler.

michael-etzkorn, 2023-01-12:
Re 1: answered here https://github.com/chipsalliance/rocket-chip/issues/2989
- blackboxing the ram would work with `$readmem`

Follow-up questions can be directed to StackOverflow or the Github discussion page.

closed 2023-01-12 by michael-etzkorn