chipsalliance/rocket-chip #1195: Creating and Testing Custom IOs for RoCC Extension

pierceja, 2018-01-16:
Hello!

I am fairly new to rocket-chip so I apologize if the answers to my questions are obvious. 

I have been messing around with the Accumulator example in the LazyRoCC.scala file. My goal is just to modify it so that it has outputs that are set when one of the custom instructions are received. I would like these outputs to go out from the RoCC extension to custom hardware that I will have inside the rocketchip. I noticed that there are two IO classes defined in this file: RoCCIO and RoCCCoreIO. I assumed that if I wanted to create outputs to go from the RoCC to custom hardware, I would define them in the RoCCIO core class. Is this correct, or does it matter?

One last question: Is there an easy way to test that the IO wires in the RoCC are being set to the correct values when I execute the custom instruction. I have had a bit of trouble using some of Chisels io testers just because the RoCC is built in with the rest of the rocketchip instead of being a stand alone module (I haven't had a lot of experience with Chisel testing so I'm unsure if I can even test the RoCC separately or not). Sorry for all the questions and I appreciate the help.

pierceja, 2018-01-30:
Sorry. I did not mean to close this issue just yet.

pierceja, 2018-02-02:
I have been looking into this whenever I get the chance. The problem I'm running into is that whenever I generate Verilog from my chisel files, the custom IOs I tried placing into the LazyRoCC.scala files are not showing up in the Verilog. They show up only after I store the IO values in a register (I think the reason they show up is because I am using them for other hardware, so I guess the interpreter picks that up when creating the Verilog files.). My goal is to use these IOs for hardware not currently on the chip, so this becomes a problem and was wondering if there was a work around. This was a hard situation to describe so I can go into more detail if needed.

seldridge, 2018-02-02:
From your description it sounds like the problem is that Firrtl is doing the right thing and ripping out your unconnected inputs.

There are a couple of ways of getting around this:
  1) Black box the thing that you want to connect the RoCC to inside of Rocket Chip and include that in the design. 
  2) Mark the signals that you want to preserve with a `DontTouchAnnotation`. There's a brief example here: https://github.com/freechipsproject/chisel3/blob/c327dc328ca819031a086ae102fefe2909831e24/chiselFrontend/src/main/scala/chisel3/core/ChiselAnnotation.scala#L34

You can verify if it is dead code elimiation (DCE) on the Firrtl side that is causing the problems by looking at the Firrtl output (*.fir) and seeing if your signals are in there. If they are and they're not in the resulting Verilog, then it's DCE being efficient. If your signals aren't there, then it's likely a problem on your end of something not doing what you want in your Chisel code.

pierceja, 2018-02-08:
Hello,

I did look into the Firrtl output and my signals were there. Applying the don't touch annotation to my signals worked and they now show up in the generated verilog files.

Thanks!

closed 2018-03-01 by hcook

SunicYosen, 2019-05-02:
> From your description it sounds like the problem is that Firrtl is doing the right thing and ripping out your unconnected inputs.
> 
> There are a couple of ways of getting around this:
> 
> 1. Black box the thing that you want to connect the RoCC to inside of Rocket Chip and include that in the design.

But all rocc examples are designed by using lazyrocc module, how can make it extend blackbox? Thanks a lot..

LIUQyou, 2021-03-12:
Hi, I encounter a problem when implementing the software design for the Rock BlackBox example. I tried to use the black box in software, and I write the code like the following. But it gives no response when I test with ./simualtion-XXX-XXXConfig  XXX.riscv in verilator.
```C
#include<stdio.h>
#include "rocc.h"

static inline void accum_write(int idx, unsigned long data)
{
	ROCC_INSTRUCTION_SS(3, data, idx, 0);
}

static inline unsigned long accum_read(int idx)
{
	unsigned long value;
	ROCC_INSTRUCTION_DSS(3, value, 0, idx, 1);
	return value;
}

static inline void accum_load(int idx, void *ptr)
{
	asm volatile ("fence");
	ROCC_INSTRUCTION_SS(3, (uintptr_t) ptr, idx, 2);
}

static inline void accum_add(int idx, unsigned long addend)
{
	ROCC_INSTRUCTION_SS(3, addend, idx, 3);
}

static inline unsigned long black_box(int num1, int num2)
{	
	unsigned long value;
	ROCC_INSTRUCTION_DSS(0, value, num1, num2, 1);
	return value;
}

unsigned long data = 0x3421L;

int main(void)
{
	unsigned long result;
	
	//Blackbox process
	black_box(1,2);
	black_box(3,4);
	result=black_box(5,6);
	printf("The result is:  %ld\n",result);
	printf("Blackbox Success!\n");
	
	accum_load(0, &data);
	accum_add(0, 2);
	result = accum_read(0);

	if (result != data + 2)
		return 1;

	accum_write(0, 3);
	accum_add(0, 1);
	result = accum_read(0);

	if (result != 4)
	{printf("Failed\n");
		return 2;}
	printf("Accum Success!\n");
	

	return 0;
}
```
But if I move the blackbox process below the Accum part, it works, which makes me very confused.

I do not where I am wrong, I am quite confused.

Could you give an example of how to use Blackbox.

Thanks in advance