chipsalliance/rocket-chip #1203: Using a proxy server in SBT

pentin-as, 2018-01-22:
Usually, for an SBT to work through a proxy server, it is advised to add an environment variable:
SBT_OPTS="-Dhttp.proxyHost=PROXYHOST -Dhttp.proxyPort=PROXYPORT -Dhttp.proxyUser=USERNAME -Dhttp.proxyPassword=XXXX -Dhttps.proxyHost=PROXYHOST -Dhttps.proxyPort=PROXYPORT -Dhttps.proxyUser=USERNAME -Dhttps.proxyPassword=XXXX $SBT_OPTS"
(or add the same thing to JAVA_OPTS).
As far as I understand, when SBT is started, this variable is passed to sbt-launch.jar somehow like this:
java $SBT_OPTS -jar `dirname $0`/sbt-launch.jar "$@"

But in rocket-chip sbt-launch.jar it is called directly, without parameter SBT_OPTS. It happens here:
https://github.com/freechipsproject/rocket-chip/blob/b8219425d8958e2bd534913e6c374c26a290d374/Makefrag#L19
and here:
https://github.com/freechipsproject/rocket-chip/blob/b8219425d8958e2bd534913e6c374c26a290d374/Makefrag#L23

I suggest adding "$(SBT_OPTS)" to these places.

edcote, 2018-03-21:
sounds like an important fix for some corporate environments (behind proxy), have you made a [pull request](https://help.github.com/articles/creating-a-pull-request/) for the community to review?

michael-etzkorn, 2023-01-11:
Proxy settings set through `JAVA_OPTS` as an environment variable are respected by SBT so if you export JAVA_OPTS that'll be sufficient to resolving this environment issue.

closed 2023-01-11 by michael-etzkorn