chipsalliance/rocket-chip #1207: Data cache miss

zhpzhp, 2018-01-24:
Is there any existing function about calculating number of cache access and cache miss?

sorear, 2018-01-24:
cache misses and load/store instructions executed can be counted by the performance monitor facility; a list of events supported by rocket-chip can currently be best found in the [SiFive documentation](https://static.dev.sifive.com/U54-MC-RVCoreIP.pdf#page=16) under §3.11 "Hardware Performance Monitor".

zhpzhp, 2018-01-26:
But how can I get access to these values in CSRs after I ran an application on the rocket chip emulator?

sorear, 2018-01-26:
add a `csrr` to the end of your application?

zhpzhp, 2018-01-26:
You mean when I compiled the code, I need to merge these `cssr` codes into the binary generated?

sorear, 2018-01-26:
No, add it to the source code

lhl2017, 2018-04-14:
Hi~ 
I try to approach to registers which are in machine mode such as minstret, mhpmcounter3-31 etc, but there is `An illegal instruction was executed!` error if I directly use 'csrr'. As I know, I have to change the mode from user's to machine's. I try to use 'mret' to change it, but I still encounter some problems. Could you please, show some sample codes for some beginners like me? I actually want to use perf counters found in the SiFive documentation under §3.11 you linked. It is really helpful to me if you can give me a hand.

Following is my sample codes.

`int main ()
{
  unsigned long instret1 = 0;
  unsigned long instret2 = 0;

  float a = 3.0;
  float b = 2.0;
  float c = 0;

  asm volatile ("csrrs %0, minstret, zero " : "=r"(instret1));
  c = a + b;
  asm volatile ("csrrs %0, minstret, zero " : "=r"(instret2));

  printf("instret count is %d \n", instret2-instret1);
  printf("c is %f \n", c);

}` 
@sorear

michael-etzkorn, 2022-05-12:
Linking #1343 and #1896. Instructing users how to take advantage of the hardware performance monitors is on the #2960 task list though it'd be nice to see a specific SO post opened about this. 

https://stackoverflow.com/questions/50317016/rocketchip-riscv-acclerator-performance-evaluation/50334126#50334126
https://stackoverflow.com/questions/50317016/rocketchip-riscv-acclerator-performance-evaluation

Perhaps a well-constructed answer to [this open question](https://stackoverflow.com/questions/50944423/how-to-monitor-performance-of-rocket-core) can be worked on to detail them more fully.

closed 2022-05-12 by michael-etzkorn