chipsalliance/rocket-chip #1210: timebase_frequency Linux

amerlo94, 2018-01-26:
Dear all,

I was trying to boot Linux on a Rocket core instantiated in a custom FPGA design and I realize that Linux will look for "timebase_frequency" in the "cpus" section of the dts, but since a while the rocket would be generated with the "timebase_frequency" parameter in the per cpu section (cpu@0,...). How linux is supposed to retrieve that data?

The function is time_init() in riscv-linux/arch/riscv/kernel/time.c.

terpstra, 2018-01-26:
The newest linux kernel looks for it in both locations. Update your kernel.

closed 2018-01-26 by terpstra

ipatix, 2019-07-05:
@terpstra I'm running into the same problem with linux 4.19.45. Is this kernel version too old?

I'd like to try a 5.X kernel, but I cannot get those to work at the moment.

terpstra, 2019-07-08:
4.9.45 should be new enough, but I don't know what patches are and are not upstream. @palmer-dabbelt

palmer-dabbelt, 2019-07-08:
Looks like we never got around to fixing that.

ipatix, 2019-07-08:
It's not critical for me since I'm already using a custom device tree which I pass from the bootrom instead of using the generated one.

Nevertheless. it would be nice to get that fixed some time in the future.

ipatix, 2019-07-17:
So I noticed setting the timebase_frequency for all cores does make Linux boot successfully. However, time is running "too slow" (a print loop with a delay of 1 second in the code takes about 2 seconds in real time) in Linux. I have a 50 MHz clock connected on my FPGA and I've set `timebase-frequency = <50000000>;` in my `cpus` block. Is that correct and if so: How do I fix it?

terpstra, 2019-07-18:
50MHz is the clock speed, but the bus speed is half that (25MHz). That said, I don't understand why it's 25MHz, since timebase-frequency is supposed to be the CLINT RTC speed, I thought, which is 1MHz???

fangrouli, 2023-06-08:
> 

Hi, can I ask in which file did you change the frequency? I tried to change it in the Configs.scala in src/main/scala/subsystem in this chunk of code:
```
case PeripheryBusKey => PeripheryBusParams(
    beatBytes = site(XLen)/8,
    blockBytes = site(CacheBlockBytes),
    dtsFrequency = Some(100000000))
```
But it does not work. The waveform generated by the verilator when I tried to run a helloworld.riscv executable is still with time period 2ps.

ZenithalHourlyRate, 2023-06-08:
> The waveform generated by the verilator when I tried to run a helloworld.riscv executable is still with time period 2ps.

The 2ps parameter is unrelated to timebase frequency.

fangrouli, 2023-06-08:
> 
> The 2ps parameter is unrelated to timebase frequency.

Hi! May I know then how should I change the clock frequency? I am thinking about adding sleep() in the emulator.cc, which can reduce the frequency, but I dont know how to increase it.

ipatix, 2023-06-08:
@fangrouli This was some time ago, but this was the device tree I used back then:
https://github.com/ipatix/rocket-chip-zedboard/blob/zedboard-untethered/bootrom/devicetree_1.dts

Look out for the lines containing `frequency`. I hope this helps.

fangrouli, 2023-06-09:
> If you want to get a real frequency, you should consider using an FPGA.

I see, indeed!

fangrouli, 2023-06-09:
> @fangrouli This was some time ago, but this was the device tree I used back then: https://github.com/ipatix/rocket-chip-zedboard/blob/zedboard-untethered/bootrom/devicetree_1.dts
> 
> Look out for the lines containing `frequency`. I hope this helps.

Thank you so much! I think it will help a lot when I progress into the FPGA simulation stage!