chipsalliance/rocket-chip #1211: About MaskROM

Xuwmzzz, 2018-01-27:
I can not build a rocket with maskrom.

I modify something and use thess configure :

modification:(rocket-chip/src/main/scala/devices/tilelink/MaskROM.scala)
case class MaskROMParams(address: BigInt = 0x10000, name: String = "BootRom", depth: Int = 2048, width: Int = 32)

configure:(rocket-chip/src/main/scala/devices/tilelink/MaskROM.scala)
class ExampleRocketSystem(implicit p: Parameters) extends RocketCoreplex
    with HasPeripheryMaskROMSlave
{
  override lazy val module = new ExampleRocketSystemModule(this)
}

class ExampleRocketSystemModule[+L <: ExampleRocketSystem](_outer: L) extends RocketCoreplexModule(_outer)
{
  val maskROMParams = p(PeriPheryMaskROMKey)
  global_reset_vector := maskROMParams(0).address.U
}

These are errors:
[info] [0.005] Elaborating design...
[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:52)
	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:49)
	at chisel3.core.Module$.do_apply(Module.scala:45)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:240)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:238)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:238)
	at chisel3.Driver$.elaborate(Driver.scala:92)
Caused by: java.lang.IllegalArgumentException: requirement failed: Key PeripheryMaskROMKey is not defined in Parameters
	at scala.Predef$.require(Predef.scala:224)
	at freechips.rocketchip.config.View.apply(Config.scala:15)
	at freechips.rocketchip.config.View.apply(Config.scala:12)
	at freechips.rocketchip.devices.tilelink.HasPeripheryMaskROMSlave$class.$init$(MaskROM.scala:17)
	at freechips.rocketchip.system.ExampleRocketSystem.<init>(ExampleRocketSystem.scala:14)
	at freechips.rocketchip.system.TestHarness$$anonfun$1.apply(TestHarness.scala:14)
	at freechips.rocketchip.system.TestHarness$$anonfun$1.apply(TestHarness.scala:14)
	at chisel3.core.Module$.do_apply(Module.scala:45)
	at freechips.rocketchip.system.TestHarness.<init>(TestHarness.scala:14)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:52)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$44$$anonfun$apply$45$$anonfun$apply$46.apply(Defaults.scala:793)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$44$$anonfun$apply$45$$anonfun$apply$46.apply(Defaults.scala:793)
	at scala.Option.foreach(Option.scala:236)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$44$$anonfun$apply$45.apply(Defaults.scala:793)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$44$$anonfun$apply$45.apply(Defaults.scala:791)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
[error] (rocketchip/compile:runMain) Nonzero exit code: 1

How can i solve it? Thank you!

terpstra, 2018-01-29:
You need to add a PeripheryMaskROMKey into your Config.

Xuwmzzz, 2018-02-01:
Thank you very much, i will try it!

Xuwmzzz, 2018-02-01:
I have a try, it is ok, thx!

edcote, 2018-03-15:
@Xuwmzzz please close the issue

closed 2018-03-16 by Xuwmzzz