chipsalliance/rocket-chip #1213: verilog is not generated

mayankagnisys, 2018-01-29:
hi,
 i got on terminal
mayank@mayank-Lenovo-B590:~/rocket-chip/vsim$ make verilog
mkdir -p /home/mayank/rocket-chip/vsim/generated-src/
cd /home/mayank/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/mayank/rocket-chip/sbt-launch.jar "runMain freechips.rocketchip.system.Generator /home/mayank/rocket-chip/vsim/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfig"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/mayank/rocket-chip/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
Using addons: 
[info] Set current project to rocketchip (in build file:/home/mayank/rocket-chip/)
[error] ## Exception when compiling 3 sources to /home/mayank/rocket-chip/chisel3/coreMacros/target/scala-2.11/classes
[error] Error opening zip file: firrtl.jar
[error] sbt.internal.inc.Locate$JarDefinesClass.liftedTree1$1(Locate.scala:78)
[error] sbt.internal.inc.Locate$JarDefinesClass.<init>(Locate.scala:75)
[error] sbt.internal.inc.Locate$.definesClass(Locate.scala:64)
[error] sbt.Defaults$.$anonfun$globalJvmCore$5(Defaults.scala:143)
[error] sbt.internal.inc.FileValueCache0$FileCache.update(FileValueCache.scala:45)
[error] sbt.internal.inc.FileValueCache0$FileCache.get(FileValueCache.scala:40)
[error] sbt.internal.inc.FileValueCache0.$anonfun$get$1(FileValueCache.scala:31)
[error] sbt.Defaults$$anon$2.definesClass(Defaults.scala:1417)
[error] sbt.internal.inc.Locate$.$anonfun$entry$1(Locate.scala:46)
[error] scala.collection.immutable.Stream.$anonfun$map$1(Stream.scala:415)
[error] scala.collection.immutable.Stream$Cons.tail(Stream.scala:1169)
[error] scala.collection.immutable.Stream$Cons.tail(Stream.scala:1159)
[error] scala.collection.immutable.StreamIterator.$anonfun$next$1(Stream.scala:1058)
[error] scala.collection.immutable.StreamIterator$LazyCell.v$lzycompute(Stream.scala:1047)
[error] scala.collection.immutable.StreamIterator$LazyCell.v(Stream.scala:1047)

jackkoenig, 2018-01-29:
Can you provide more information about your environment: OS, java version, etc.?

terpstra, 2018-01-29:
This error is caused by a change in the sbt version. You need to blow away all directories named target/ and it will work.

mayankagnisys, 2018-01-30:
hi terpstra,thanks for response
      i blew all directories away named target/
i got same
[info] Compiling 3 Scala sources to /home/mayank/rocket-chip/chisel3/coreMacros/target/scala-2.11/classes ...
[error] error while loading <root>, Error accessing /home/mayank/rocket-chip/lib/firrtl.jar
[error] ## Exception when compiling 3 sources to /home/mayank/rocket-chip/chisel3/coreMacros/target/scala-2.11/classes
[error] object java.lang.Object in compiler mirror not found.
[error] scala.reflect.internal.MissingRequirementError$.signal(MissingRequirementError.scala:17)
[error] scala.reflect.internal.MissingRequirementError$.notFound(MissingRequirementError.scala:18)
[error] scala.reflect.internal.Mirrors$RootsBase.getModuleOrClass(Mirrors.scala:53)
[error] scala.reflect.internal.Mirrors$RootsBase.getModuleOrClass(Mirrors.scala:45)
[error] scala.reflect.internal.Mirrors$RootsBase.getModuleOrClass(Mirrors.scala:45)
---------------
---------------

mayankagnisys, 2018-01-30:
hi jackkoenig
  os- Ubuntu 16.04.3 LTS
java-jdk8.0

mayankagnisys, 2018-01-30:
i again do same processes only execute make verilog .issue has been resolved,thanks
but i didnot get alu.v and others like scala to verilog rtl conversion.can we get alu.v from alu.scala from src folder of rocket-chip repository?

jackkoenig, 2018-02-06:
There is no 1:1 correspondence between Chisel (Scala) files and Verilog.

If you run `make verilog` in `rocket-chip/vsim` you'll get the emitted Verilog file `rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.v`. Inside of that file you will find `module ALU` that corresponds to [class ALU](https://github.com/freechipsproject/rocket-chip/blob/d78ad857ee967946453d2c1500faf2a6e3d82d2e/src/main/scala/rocket/ALU.scala#L50) in `rocket-chip/src/main/scala/rocket/ALU.scala`. 

If you would like to generate one Verilog file per module, you can change the [firrtl command-line options](https://github.com/freechipsproject/rocket-chip/blob/c1eb795abafc3774c69560bbe3b0cff5210bdd21/vsim/Makefrag-verilog#L17) like so:
`$(FIRRTL) -i $< -tn $(MODEL) -td $(generated_dir) -X verilog --infer-rw $(MODEL) --repl-seq-mem -c:$(MODEL):-o:$(generated_dir)/$*.conf -faf $(generated_dir)/$*.anno --split-modules`

Note that while this change will not work with the rest of rocket-chip's flow.