chipsalliance/rocket-chip #1217: using blackbox in the rocket-chip repo

allencho1222, 2018-01-31:
I'm trying to use BlackBox in rocket-chip generator.

First of all, I tried it using chisel3 template, and it works well.

However, I tried it inside rocket-chip generator using setResourcee("/alu/customALU.v") in the class that extends BlackBox with HasBlackBoxResource.

The location of file:  "~/rocket-chip/src/main/resources/alu/customALU.v"

But it continues to show FileNotFoundException:

mkdir -p /home/ubuntu-cho/rocket-chip/emulator/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/ubuntu-cho/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/ubuntu-cho/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.fir -o /home/ubuntu-cho/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.v -X verilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/home/ubuntu-cho/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.conf -faf /home/ubuntu-cho/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.anno -ffaaf
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
**Exception in thread "main" java.io.FileNotFoundException: Resource '/rocket/customALU.v'**
        at firrtl.transforms.BlackBoxSourceHelper$.copyResourceToFile(BlackBoxSourceHelper.scala:145)
        at firrtl.transforms.BlackBoxSourceHelper$$anonfun$1.apply(BlackBoxSourceHelper.scala:107)
        at firrtl.transforms.BlackBoxSourceHelper$$anonfun$1.apply(BlackBoxSourceHelper.scala:103)
        at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at firrtl.transforms.BlackBoxSourceHelper.execute(BlackBoxSourceHelper.scala:103)
        at firrtl.Transform$$anonfun$4.apply(Compiler.scala:242)
        at firrtl.Transform$$anonfun$4.apply(Compiler.scala:242)
        at firrtl.Utils$.time(Utils.scala:135)
        at firrtl.Transform.runTransform(Compiler.scala:242)
        at firrtl.Compiler$$anonfun$13$$anonfun$apply$10.apply(Compiler.scala:461)
        at firrtl.Compiler$$anonfun$13$$anonfun$apply$10.apply(Compiler.scala:461)
        at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
        at scala.collection.immutable.List.foldLeft(List.scala:84)
        at firrtl.Compiler$$anonfun$13.apply(Compiler.scala:461)
        at firrtl.Compiler$$anonfun$13.apply(Compiler.scala:461)
        at firrtl.Utils$.time(Utils.scala:135)
        at firrtl.Compiler$class.compile(Compiler.scala:460)
        at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:138)
        at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:176)
        at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:140)
        at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:129)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at logger.Logger$.makeScope(Logger.scala:127)
        at firrtl.Driver$.execute(Driver.scala:140)
        at firrtl.Driver$.execute(Driver.scala:232)
        at firrtl.Driver$.main(Driver.scala:248)
        at firrtl.Driver.main(Driver.scala)
Makefrag-verilator:16: recipe for target '/home/ubuntu-cho/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.v' failed
make: *** [/home/ubuntu-cho/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.v] Error 1

allencho1222, 2018-01-31:
I found that verilog file used for BlackBox have to exist inside "~/rocket-chip/firrtl/src/main/resources".
Why this file isn't managed by directory, "~/rocket-chip/src/main/resources"? If it is possible, what should be modified?

grebe, 2018-02-02:
I played around with doing this here: grebe/rocket-chip/pull/1

I'm going to PR something like this soon.

michael-etzkorn, 2023-01-12:
Fixed by #1302.

closed 2023-01-12 by michael-etzkorn