chipsalliance/rocket-chip #1220: Emulator compiling failed

charlesnieh, 2018-02-01:
While doing the compile in emulator directory, the following error stops the compilation:

make VM_PARALLEL_BUILDS=1 -C /home/user/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig -f VTestHarness.mk
make[1]: Entering directory '/home/user/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig'
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -Wno-char-subscripts -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -O1 -std=c++11 -I/home/user/rocket-chip/include -I/home/ycnieh/rocket-chip/csrc -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/user/rocket-chip/csrc/verilator.h -I/home/user/rocket-chip/emulator/generated-src -include /home/user/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig/VTestHarness.h   -c -o SimDTM.o /home/user/rocket-chip/csrc/SimDTM.cc
/home/user/rocket-chip/csrc/SimDTM.cc: In function ‘int debug_tick(unsigned char*, unsigned char, int*, int*, int*, unsigned char, unsigned char*, int, int)’:
/home/user/rocket-chip/csrc/SimDTM.cc:26:43: error: no matching function for call to ‘dtm_t::dtm_t(PLI_INT32&, PLI_BYTE8**&)’
       dtm = new dtm_t(info.argc, info.argv);
                                           ^
In file included from /home/user/rocket-chip/csrc/SimDTM.cc:3:0:
/home/user/rocket-chip/include/fesvr/dtm.h:17:3: note: candidate: dtm_t::dtm_t(const std::vector<std::__cxx11::basic_string<char> >&)
   dtm_t(const std::vector<std::string>& args);
   ^~~~~
/home/user/rocket-chip/include/fesvr/dtm.h:17:3: note:   candidate expects 1 argument, 2 provided
VTestHarness.mk:71: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '/home/user/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig'
Makefrag-verilator:71: recipe for target 'emulator-freechips.rocketchip.system-DefaultConfig' failed
make: *** [emulator-freechips.rocketchip.system-DefaultConfig] Error 2

I am on the master branch now. The compilation didn't fail in previous version. Any idea?

seldridge, 2018-02-01:
It looks like you have an older version of riscv-fesvr installed. 

There was [an API change to riscv-fesvr](https://github.com/riscv/riscv-fesvr/pull/25) that it looks like you don't have.

This should clear up if you bump riscv-tools and rebuild:
```bash
git submodule update --init --recursive
cd riscv-tools
./build.sh
```

_Note: you may be able to get away with only rebuilding and installing riscv-fesvr as opposed to rebuilding the entire toolchain._

charlesnieh, 2018-02-01:
@seldridge Many thanks, it works magically.

closed 2018-02-01 by charlesnieh