chipsalliance/rocket-chip #1225: Add "shadow" peripherals

amerlo94, 2018-02-05:
Dear all,

I've instantiated the Rocket core in a custom design where a bunch of peripherals are mapped over the MMIO region. I am running Linux too, thus I would like to add these peripherals into the DTB, so bbl could scan them and give them to the Linux kernel. My question is: "How do I add a shadow peripheral in Scala in order to show up in the autogenerated dts located in the BootROM?"

edcote, 2018-02-07:
Have no idea what you mean by "shadow" peripheral.

The following may be helpful: http://www.lowrisc.org/docs/diplomacy/.

amerlo94, 2018-02-08:
Yeah, this is exactly the issue that I am facing. Thanks for the suggestion!

amerlo94, 2018-02-08:
Since the post has been written by @wsong83, is there any resources on the changes in the Diplomacy package?

amerlo94, 2018-02-13:
I look trough the VirtualNode code made by lowRISC, but since there are some changes in how the Diplomacy package works, and in particular in commit 9217baf9d415c06fe17d96dffa72f5a05ac879ac it seems to be an auto-generated code for the node bundle. Is there any source where i could find more info about it?

michael-etzkorn, 2022-06-04:
Part of the documentation plan includes adding documentation for Diplomacy and the DeviceTree as diplomacy is split out to help answer this flavor of question. For now, due to age, I'll close the issue and encourage a repost to Stack Overflow if you're interested in following up on these questions still.

closed 2022-06-04 by michael-etzkorn