chipsalliance/rocket-chip #1227: About the macro definiton

Xuwmzzz, 2018-02-05:
Dear all,

I find that the VCS simulation of the rocket core must use some macro definitons, like these:
	+define+RANDOMIZE_MEM_INIT \
	+define+RANDOMIZE_REG_INIT \
	+define+RANDOMIZE_GARBAGE_ASSIGN \
	+define+RANDOMIZE_INVALID_ASSIGN \

Now, i want to mapping a rocket core down to an FPGA. If i use these macro definitons, some syntax can not be synthesised, like this:

        `ifdef RANDOMIZE
            integer initvar;
            initial begin
                `ifndef verilator
                    #0.002 begin end
                `endif
                `ifdef RANDOMIZE_REG_INIT
                _RAND_0 = {1{$random}};
                _T_821 = _RAND_0[2:0];
                `endif
        `endif

I can not synthesis, so i can not map the core down to the FPGA, how can i solve it? Thank you!

Xuwmzzz, 2018-02-05:
By the way, i can not get the waves of the rocket core if i do not use these macro definitons.

+define+RANDOMIZE_MEM_INIT 
+define+RANDOMIZE_REG_INIT 
+define+RANDOMIZE_GARBAGE_ASSIGN 
+define+RANDOMIZE_INVALID_ASSIGN

edcote, 2018-02-07:
To your first comment.  You are correct.  You cannot synthesize an initial block.  The intent for these preprocessor defines is for functional simulation so as to eliminate possible sources of "X's" and help root cause any issues with X propagation.

I cannot answer your second question.  I suspect you are doing something wrong.

Xuwmzzz, 2018-02-08:
Are those registers with the value of "X's" be pruned during the process of syntiesis? and will the function of processor be influenced by those synthesized registers? 

I reset some registers and get the waves of the rocket core without those macro definitons by VCS.
I think the problem is cause by the value of "X's", so i am not sure the rocket can work after synthesizing.

edcote, 2018-02-08:
Happy debugging.  You are facing a very common verification problem.

More background: http://www.techdesignforums.com/practice/guides/x-propagation/

Xuwmzzz, 2018-03-01:
OK, thank you very much.

edcote, 2018-03-21:
@Xuwmzzz please close

closed 2018-03-22 by Xuwmzzz

STAIRB, 2018-03-29:
Hi everyone,

I am interested in this problem since I want to synthesize it on FPGA but when RANDOMIZE_GARBAGE_ASSIGN is not defined there are wires like metaArb_io_in_4_bits_addr that are not driven. 
Thus I would like to know what should these wires(metaArb_io_in_4_bits_addr ) be assigned to when RANDOMIZE_GARBAGE_ASSIGN is not defined, is 0 or 1 or something else ?
and also, does anyone know the purpose of metaArb_io_in_4_bits_addr.

Thank you all