chipsalliance/rocket-chip #1233: Bring RoCC pins to system

kenzhang82, 2018-02-08:
Hi there,

I am just wondering if there is any way to bring out the RoCC pins out to system in this version of rocket chip (https://github.com/freechipsproject/rocket-chip/tree/7e75d63ba6b4c1b50aaaf920e1c693ef6acf51d7)? It seems like the RoCC is wrapped by some kind of clock crossing wrapper (see: https://github.com/freechipsproject/rocket-chip/blob/7e75d63ba6b4c1b50aaaf920e1c693ef6acf51d7/src/main/scala/coreplex/RocketCoreplex.scala#L89), I am not too sure how to access (retrieve) from system level. I didn't see any existing example of bringing a pin from tile to system.

Any help would be much appreciated! Thanks!

terpstra, 2018-02-08:
Just like regular Verilog you will need to add an IO to each Module level to plumb the wires out.
Something along the lines of:
val extra_io = IO(Vec(tiles.map(_.myExtraIO)))
(extra_io zip tiles.map(_.myExtraIO)) foreach { case (a,b) => a := b }

kenzhang82, 2018-02-09:
Thanks @terpstra  and @seldridge!

I managed to plumb the wires out up to RocketTileWrapper level, I could see my signals in fir (if that is the right way to check?) but not further.

When I apply your "chisel magic" 

val extra_io = IO(Vec(tiles.map(.myExtraIO)))

to here (https://github.com/freechipsproject/rocket-chip/blob/7e75d63ba6b4c1b50aaaf920e1c693ef6acf51d7/src/main/scala/coreplex/RocketCoreplex.scala#L86), I got an error saying "not found: value IO".

Also, from system level, I have no way to access the signal, I've already tried sys.wrapper but this steps into the wrong bundle..

codelec, 2018-02-16:
@kenzhang82  you have to do it here https://github.com/freechipsproject/rocket-chip/blob/7e75d63ba6b4c1b50aaaf920e1c693ef6acf51d7/src/main/scala/coreplex/RocketCoreplex.scala#L140 
IO should be used in ModuleImp

yjwen, 2018-05-04:
For anyone who is also interested in this topic, I managed to route RoCC IOs to the system, by the following steps.

1. Add extra IO in `LazyRoCCModule` by inserting
```scala
val xio = IO(new RoCCXIO)
```
at https://github.com/freechipsproject/rocket-chip/blob/d4228836f9855167350427aa9ba3b0e3d75062d4/src/main/scala/tile/LazyRoCC.scala#L73

2. Add extra IO in `RocketTileModuleImp` by inserting
```scala
  val rocc_xio = IO(new RoCCXIO)
  if (outer.roccs.length > 0)
    rocc_xio := outer.roccs(0).module.xio
```
at https://github.com/freechipsproject/rocket-chip/blob/d4228836f9855167350427aa9ba3b0e3d75062d4/src/main/scala/tile/RocketTile.scala#L150

3. Add extra IO in `RocketSub\systemModuleImp` by inserting
```scala
  val rocc_xio = IO(new RoCCXIO)
  if (_outer.rocketTiles.length > 0)
    rocc_xio := _outer.rocketTiles(0).module.rocc_xio
```
at https://github.com/freechipsproject/rocket-chip/blob/d4228836f9855167350427aa9ba3b0e3d75062d4/src/main/scala/subsystem/RocketSubsystem.scala#L162

4. Modify TestHarness.scala to response to RoCC requests.

kenzhang82, 2018-05-08:
Thanks @yjwen! That helps, but have you tried routing the pins one level up further? (aka. platform in sifive repo?)

kenzhang82, 2018-05-09:
@seldridge, do you have any idea? I can route it all the way up to system, but not to the top. Would be good if you can give me some pointers. :)

aicr, 2018-05-21:
@yjwen  step 4 of your explanation is a little bit vague. can you elaborate ?

aicr, 2018-05-23:
Is it somehow possible to add the new extra io bundle to your own  roccModuleImp class , the one that extends `LazyRoCCModuleImp` ?

The problem I have with adding it to `LazyRoCCModuleImp` is that you add also to all other RoCC as well.
 
I tried that and get the next error (fxio is my  extra ios) :

> value fxio is not a member of freechips.rocketchip.tile.LazyRoCCModuleImp