chipsalliance/rocket-chip #1237: Wiring ExtInterrupts to the Plic

amerlo94, 2018-02-14:
Hey,
I am using external interrupts provided by the trait HasAsyncExtInterrupts, but I am not able to make the actual wiring from the external interrupts pins to the Xing, Xbar and then to the Plic.

In the generated Verilog file I discover that the input of both the Plic and the related Xing are hardwired to 0 for all the external interrupts.

I think I am missing something also looking at this comment in the InterruptBus.scala file. Where I am supposed to make the actual connections?
```
/** This trait performs the translation from a UInt IO into Diplomatic Interrupts.
  * The wiring must be done in the concrete LazyModuleImp. 
  */
trait HasExtInterruptsModuleImp extends LazyModuleImp with HasExtInterruptsBundle {
  val outer: HasExtInterrupts
  val interrupts = IO(UInt(INPUT, width = outer.nExtInterrupts))

  outer.extInterrupts.out.map(_._1).flatten.zipWithIndex.foreach { case(o, i) => o <> interrupts(i) }
}
```

codelec, 2018-02-14:
Can you post a link to your modified code because it is difficult to interpret from the information that you have given what the problem might be

amerlo94, 2018-02-14:
I have not changed any code for the extInterrupts, I use:

`  case NExtTopInterrupts => 16 // from 31 in the BaseConfig`

 and

```    
with HasAsyncExtInterrupts
with HasExtInterruptsModuleImp
```

in the Top design.

This should be enough, right? I have the vector of interrupts as output of my system, but internally the input of the Xing are tied to '0', such as the input of the Xbar.

codelec, 2018-02-20:
i am not able to reproduce that. can you post snippets of your firrtl

amerlo94, 2018-02-21:
Thanks but i solved the issue, I was using an old version of the repo without the "Don't Touch" package.

closed 2018-02-21 by amerlo94