chipsalliance/rocket-chip #1255: [Question] How to configure different clock frequency between systembus and periphery bus?

RaulGoal, 2018-02-28:
Hi,
I have added a simple periphery module on periphery bus and executed c simulation.
I intend to make clock frequency of periphery bus distinguish from system bus.

I try and modify below  frequency value to 5000000 for achieving my goal.
https://github.com/freechipsproject/rocket-chip/blob/a48dd575b23aa4b82368f19beaaf19e2d4bf1e3e/src/main/scala/coreplex/PeripheryBus.scala#L18

 I have checked the simulation wave form diagram after such modification. 
[image]
However, periphery bus clock frequency remains same as system bus.

I think that I use wrong approach to achieve my goal.
Can anyone tells me how to set it or the method?

RaulGoal, 2018-03-01:
Thanks for your reply.
I have looked the csrc/emulator.cc file. It probably have to generate the othe clock as `tile->clk`  for periphery bus. However, it might also need to some modifications or addition to Chisel implement code
for identifying the module work at clk frequency.
ig:
`class PeripheryBus(params: PeripheryBusParams)(implicit p: Parameters) extends TLBusWrapper(params, "PeripheryBus") {                                                                                                                                            `
 `clock :=clk  //secify clock frquency = clk`
 `...........}`

I am not familiar with such modification.
How to do that?
Can you give me some suggestions or tips?

terpstra, 2018-03-01:
Just FYI, no one has done what you are trying to do and it's highly unlikely we would merge changes to this effect back into emulator.cc. That said, your best bet is probably to move the tiles into their own clock domain and plumb signals out of the design for them. Shoving the Periphery is something I know @hcook is working to make easier, but AFAIK this work is not completed.

hcook, 2018-03-01:
#1190 adds support for specifying a clock crossing between the system bus and periphery bus, but doesn't actually create a different clock wired to the outside. You could do that manually. The hard part is that then all the slave devices attached to the periphery bus had better be put on its same clock... I think we'll have better support for this in a month or two.

codelec, 2018-03-01:
so am i correct on this that periphery bus means bus supporting only TL-UL type of transactions and not an under-clocked bus for slower peripherals?

GitZgl, 2020-03-06:
I am aslo confused that how can I generate a clock crossing SOC. In this Soc, system bus running on a fast clock, and periphery bus running on a slow clock.