chipsalliance/rocket-chip #1257: Can the data transmit from L2 to core without the dcache?

Xuwmzzz, 2018-03-01:
Dear all,

I found that the data be transmitted from L2 through the dcache to core.
How about transmitting from L2 to core without the dcache? 
If possible, how can i configure the rocket-chip by the scala code.

Thank you!

terpstra, 2018-03-01:
I do not understand this question. There is no L2 in rocket-chip. Furthermore, even if there were, it would only deliver data to the core's L1D, never directly to the core.

Xuwmzzz, 2018-03-01:
I mean that can the core extract data directly from external memory/L2? Not from the L1D.

Xuwmzzz, 2018-03-01:
Or can I modify the value of some registers to achieve this effect?

terpstra, 2018-03-01:
Do you mean MMIO?

Xuwmzzz, 2018-03-02:
No, i mean that can the Dcache be used selectively when the program is running?

closed 2018-04-04 by Xuwmzzz