chipsalliance/rocket-chip #1258: Why there is tag bits in rocc memory interface ?

abdullahrazakhan91, 2018-03-01:
Hi
Tag bits in rocc memory interface is used collect data from data cache with respect to same address. i m just confused why there is tag bits in rocc memory interface. can we not get tag from address. ?

thanks

hcook, 2018-03-01:
"Tag" in this case refers generically to the transaction identifier used by the processor (or RoCC) to identify which outstanding request to the data cache is now having its response sent back. I agree the overloaded use of the word when applied to a cache interface is confusing...

seldridge, 2018-03-07:
It is entirely up to the user (the RoCC) to set the tags. You can set them to part of the address or you can set them to arbitrary things which make it easier to build the RoCC (like an `id` of the internal requesting unit, a fixed value, or to some portion of the address). The value of the tag shouldn't have an effect on how the system behaves outside of your RoCC, however, only that the tag will be returned with the request.

closed 2018-03-08 by hcook