chipsalliance/rocket-chip #1265: Some doubts about TileLink interface implementation

nnamenoname, 2018-03-04:
Hello, I am a beginner of rocket chip and sorry if my question is silly.
(1) TileLink can support burst transfer. 
ig: start address = 0x0000
     transfer size = 128 byte
Address is increased 8 byte in every beats. Where can I find such implementation with corresponding code?
What is the function of TLWidthWidget module?  I find the similar process  is in WidthWidget.scala, but I am not sure whether it does address increment.
(2)How can I wrap two master port within a single module?
I find the code src/main/scala/tilelink/PatternPusher.scala. It seems like simple but not complete example.
And it wrap a single master port in it.
https://github.com/freechipsproject/rocket-chip/blob/20a88768560ca810c0dbe79f1a33604268d921c7/src/main/scala/tilelink/PatternPusher.scala#L38
https://github.com/freechipsproject/rocket-chip/blob/20a88768560ca810c0dbe79f1a33604268d921c7/src/main/scala/tilelink/PatternPusher.scala#L46
Is it like the below?
class TLPatternPusher{
  val node1 = TLClientNode(...)
   val node2 = TLClientNode(...) 

   val (tl_out1, edgeOut) = node1.out(0)
   val (tl_out2, edgeOut) = node2.out(0)

   tl_out1 := something
   tl_out2 := somthing   
}

codelec, 2018-03-04:
1. you can try using firstlastHelper or friends in tilelink/Edges.scala
TLWidthWidget is used as adapter between client and manager node if they support different  beatBytes

2. 
```scala 
class TLPatternPusher{
val node1 = TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters(name = name))),TLClientPortParameters(Seq(TLClientParameters(name = name2)))))
lazy val module = new LazyModuleImp(this) {
val (tl_out, edgeOut) = node1.out(0)
val (tl_out1, edgeOut1) = node1.out(1)

tl_out1 := something
tl_out2 := somthing
}
}
```

nnamenoname, 2018-03-04:
Thanks for your reply.
Advance asking about question 2: 
How can I attach separate master ports on bus?
Is it like the below? 
ig: one for system bus, and the other for periphery bus
```
sbus.fromSyncPorts()   := TLPatternPusher.node(0)
pbus.fromSyncPorts()   := TLPatternPusher.node(1)
```
Or in other way?

codelec, 2018-03-04:
looking at the following code from diplomacy/Nodes.scala 
```scala
class SourceNode[D, U, EO, EI, B <: Data](imp: NodeImp[D, U, EO, EI, B])(po: Seq[D])(implicit valName: ValName)
  extends MixedNode(imp, imp)
{
  protected[diplomacy] def resolveStar(iKnown: Int, oKnown: Int, iStars: Int, oStars: Int): (Int, Int) = {
    require (oStars <= 1, s"${name} (a source) appears right of a :=* ${oStars} times; at most once is allowed${lazyModule.line}")
    require (iStars == 0, s"${name} (a source) cannot appear left of a :*=${lazyModule.line}")
    require (iKnown == 0, s"${name} (a source) cannot appear left of a :=${lazyModule.line}")
    require (po.size == oKnown || oStars == 1, s"${name} (a source) has only ${oKnown} outputs connected out of ${po.size}")
    require (po.size >= oKnown, s"${name} (a source) has ${oKnown} outputs out of ${po.size}; cannot assign ${po.size - oKnown} edges to resolve :=*${lazyModule.line}")
    (0, po.size - oKnown)
  }
  protected[diplomacy] def mapParamsD(n: Int, p: Seq[D]): Seq[D] = po
  protected[diplomacy] def mapParamsU(n: Int, p: Seq[U]): Seq[U] = Seq()
}
```
It should be following (sorry I don't have any example tilelink node at hand with which i can try this to confirm)
1. 
```scala 
sbus.fromSyncPorts()   := TLPatternPusher.node
pbus.fromSyncPorts()   := TLPatternPusher.node
```
2.
```scala
sbus.fromSyncPorts()   :=* TLPatternPusher.node
```

nnamenoname, 2018-03-05:
I have tried the approaches stated above with simple example .
However, it fails working.
Here is my partial example code. (ignore the function of the example module)
```scala
class TLMasterTest (address: BigInt, beatBytes: Int)(implicit p: Parameters) extends LazyModule {
 
  val node = TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters(
    sourceId = IdRange(0, 1), 
    name = s" TileLink Test")))))

  lazy val module = new LazyModuleImp(this) {

   val (tl, edge)   = node.out(0)    
   val (tlp, edgep) = node.out(1)
    
    val read_data = edge.Get(
      fromSource = 0.U,
      toAddress = read_from_addr,
      lgSize = 3.U)._2

    val write_data = edgep.Put(
      fromSource = 0.U,
      toAddress = write_back_addr,
      lgSize = 3.U,
      data = accumulator)._2
  
    tl.a.valid := (state === mem_read_req)
    tl.d.ready := true.B
    tl.a.bits  := read_data

    
    tlp.a.valid := (state === mem_write) 
    tlp.d.ready := true.B
    tlp.a.bits  := write_data

    //unusage signal
    tl.b.ready := false.B
    tl.c.valid := false.B
    tl.e.valid := false.B

    tlp.b.ready := false.B
    tlp.c.valid := false.B
    tlp.e.valid := false.B

    .....
    memory read or write processing function
    ....

}
```

And it fails at compiling stages.
select error message:
```
[0m[[0m[31merror[0m] [0m[0m	at LongSimplewithMaster.TLMasterTest$$anon$1.<init>(TLMasterTest.scala:34)[0m
```

code for 34 row in my code
```sacala
val (tlp, edgep) = node.out(1)
```
It seems it needs the other methods.

codelec, 2018-03-05:
```scala
TLClientNode(Seq(TLClientPortParameters(Seq(TLClientParameters(
    sourceId = IdRange(0, 1), 
    name = s" TileLink Test")))))
```
you have to add another TLClientPortParameters to Seq

nnamenoname, 2018-03-05:
Sorry I can not understand it.
Could you give me more concrete example?

hcook, 2018-03-08:
It is forbidden, because atomic operations are supported for all devices on `pbus` via the `AutomicAutomata` instantiated between the `sbus` and `pbus`. Having a single master is necessary to enforce atomicity. Attach to the `sbus` instead.

nnamenoname, 2018-03-09:
I get it. 
Thanks for your suggestion.

closed 2018-03-09 by nnamenoname