chipsalliance/rocket-chip #1269: sbt encounter unresolved dependency when use make in emulator

rsthnn, 2018-03-06:
Hi. 
I have tried several times to run "make" under rocket-chip/emulator. I have cloned the latest version in 2018/3/2.  I have clone the project and done each step just as what the README.md saied,  but I meet such kind of issues like below:
make -C /home/riscv/work/rocket-chip/firrtl SBT="java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/riscv/work/rocket-chip/sbt-launch.jar" root_dir=/home/riscv/work/rocket-chip/firrtl build-scala
make[1]: Entering directory '/home/riscv/work/rocket-chip/firrtl'
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/riscv/work/rocket-chip/sbt-launch.jar "assembly"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/riscv/work/rocket-chip/firrtl/project
[info] Updating ProjectRef(uri("file:/home/riscv/work/rocket-chip/firrtl/project/"), "firrtl-build")...
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: org.eclipse.jgit#org.eclipse.jgit;4.5.0.201609210915-r: Resolution failed several times for dependency: org.eclipse.jgit#org.eclipse.jgit;4.5.0.201609210915-r {compile=[default(compile)]}:: 
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit/4.5.0.201609210915-r/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	Resolution failed several times for dependency: org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r {}:: 
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/jars/org.eclipse.jgit-parent.jar: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/jars/org.eclipse.jgit-parent.jar: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/jars/org.eclipse.jgit-parent.jar: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	sonatype-releases: unable to get resource for org/eclipse/jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://oss.sonatype.org/content/repositories/releases/org/eclipse/jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/org.eclipse.jgit-parent-4.5.0.201609210915-r.pom: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	sonatype-snapshots: unable to get resource for org/eclipse/jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://oss.sonatype.org/content/repositories/snapshots/org/eclipse/jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/org.eclipse.jgit-parent-4.5.0.201609210915-r.pom: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 
[warn] 	Resolution failed several times for dependency: org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r {}:: 
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/jars/org.eclipse.jgit-parent.jar: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/jars/org.eclipse.jgit-parent.jar: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/jars/org.eclipse.jgit-parent.jar: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	sonatype-releases: unable to get resource for org/eclipse/jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://oss.sonatype.org/content/repositories/releases/org/eclipse/jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/org.eclipse.jgit-parent-4.5.0.201609210915-r.pom: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	sonatype-snapshots: unable to get resource for org/eclipse/jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://oss.sonatype.org/content/repositories/snapshots/org/eclipse/jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/org.eclipse.jgit-parent-4.5.0.201609210915-r.pom: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 
[warn] 	Resolution failed several times for dependency: org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r {}:: 
[warn] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun [... cut ...]
[error] 	typesafe-ivy-releases: unable to get resource for org.eclipse.jgit#org.eclipse.jgit-parent;4.5.0.201609210915-r: res=https://repo.typesafe.com/typesafe/ivy-releases/org.eclipse.jgit/org.eclipse.jgit-parent/4.5.0.201609210915-r/jars/org.eclipse.jgit-parent.jar: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
... ...

Can you help to give some guide to debug this issues?   We have been sufferring this issue for 2 weeks.

edcote, 2018-03-15:
This issue is not related to the rocketchip repository.  Try stackoverflow

https://stackoverflow.com/search?q=+PKIX+path+building+failed

zhulangpi, 2018-04-21:
I have solved this question by installing chisel3 separately. Y can try it as follows.

https://github.com/freechipsproject/chisel3#installation

rsthnn, 2018-05-15:
Thanks all for giving some suggestion for my issue.  After several trials, I found the issues is that I can't access the default repositories, so I have resolve it by following steps:
1.  create file named "respository", and copy the following content into it:
[repositories]
  local
  aliyun: http://maven.aliyun.com/nexus/content/groups/public/
  sbt-plugins-repo: http://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/, [organization]/[module]/(scala_[scalaVersion]/)(sbt_[sbtVersion]/)[revision]/[type]s/[artifact](-[classifier]).[ext]
  typesafe-sbt-plugins: http://dl.bintray.com/sbt/sbt-plugin-releases/, [organization]/[module]/(scala_[scalaVersion]/)(sbt_[sbtVersion]/)[revision]/[type]s/[artifact](-[classifier]).[ext], bootOnly
  central: http://repo1.maven.org/maven2/

2.  add "-Dsbt.override.build.repos=true"  into file "/usr/share/sbt/conf/sbtopts".   This is the key steps to make the flow use the new created repository.

3.  run "sbt compile" under rocket-chip

4.  under rocket-chip/firrtl, run:
       sbt publishLocal        
       sbt compile
5.  run "make " under rocket-chip/emulator

closed 2018-05-15 by rsthnn

SunYongshuai, 2018-10-31:
I have the same problem when I use default 'sbt.version=1.1.1', but when I change to the newest 'sbt.version=1.2.6', this error disappear. [Already add repositories in $HOME/.sbt/]
It still doesn't work when I do as @rsthnn suggest.
Thanks a lot If you have any suggestions.