chipsalliance/rocket-chip #1272: Exceptions for non-cached regions

mayoru, 2018-03-07:
I see that no exceptions (instruction address fault or load access fault) generated when addressing to a non-cached region (SimAXIMMIO at mmio_axi4 bus in TestHarness) as opposed to a cached region (SimAXIMem at mem_axi4 bus in TestHarness)
Is that really intended behavior?

mayoru, 2018-03-09:
I intend to replace the memories in TestHarness with my own devices, which are able to return SLVERR or DECERR on AXI.
For now I forced AXI RRESP[1] to 1 in the simulator to generate SLVERR and catch an exception.
I see access fault signals (*_ae_ld) coming into the core set to 1 when reading (LWing or JUMPing) from mem_axi4 and no signals set when reading from mmio_axi4 bus. In both cases the error signal on TLbus is set, though.
I wonder if no device can generate access fault if placed on mmio_axi4 bus?

aswaterman, 2018-03-09:
I see.

Errors on I/O accesses cause imprecise interrupts, not synchronous exceptions.  This is because the I/O accesses commit, and the processor continues executing younger instructions, before the error condition is known.

If you want to detect these imprecise interrupts, you can enable the bus-error unit (https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/rocket/BusErrorUnit.scala).  This unit has some programmable control registers that indicate which events cause interrupts.  You can enable it by setting ```RocketCoreParams.tileControlAddr``` to an address at which to instantiate the bus-error unit's control registers.  Unfortunately this feature has no documentation at this time.

mayoru, 2018-03-26:
Thank you, aswaterman.

closed 2018-04-10 by hcook