chipsalliance/rocket-chip #1277: Can rocket-chip be run on FPGA-only FPGA?

usnull0000, 2018-03-09:
I don't know how to call FPGAs like VIRTEX-6, which doesn't have ARM SoC part.
Can this be synthesized and connected to host PC through UART serial terminal on VIRTEX-6(https://www.xilinx.com/products/boards-and-kits/dk-v6-embd-g.html) development board?

CLappin, 2018-03-09:
Short answer is yes, the rocket chip can be synthesized for FPGA only boards.  I haven't used it on the Virtex-6 but have generated and used it on Microsemi's Creative Igloo2 board which is FPGA only.  

For my set-up I had the Rocket-chip connected to DDR/LSRAM on 0x8000_0000, peripherals (GPIO,UART etc) on 0x7000_0000 and eNVM on 0x6000_0000. 

I'm unsure if there are any Xilinx specific things to look out for though.

mayoru, 2018-04-10:
I personally use Xilinx Zynq 7000, but my assembly has a JTAG debug connector.

STAIRB, 2018-04-16:
@CLappin I am trying to connect the rocket chip on an FPGA without an ARM SOC, could you please elaborate on how did you connected the Rocket Chip with the mentioned peripherals. I am fairly new to building SOCs and looked a lot for an example on how to add a peripheral to the rocket chip but I couldn't find anything except  how to configure the rocket chip with the chisel parameters.

CLappin, 2018-04-16:
Hi @STAIRB , Including WithJtagDTM in your core will give you JTAG  signals in your ExampleRocketSystem module. they will be called debug_systemjtag_jtag_* ( * = TCK, TMS,TDI,TDO_data, TDO_driven, reset, mfr_id etc). The main signals that you have to look at are TDI, TDO_data, TMS, reset (TRST). These are the main JTAG signals.  You can then attach these signals up to pins on your board and use a debugger, such as the  [olimex-arm-tiny-usb](https://www.olimex.com/Products/ARM/JTAG/ARM-USB-TINY-H/). You will need to use jumper cables from the debugger to the board as the included cable won't work for this implementation. This is just one implementation that I have seen working.

As an aside, as Chisel is based on Scala it would be a good idea to look into how Scala works as a lot of the functionality is from Scala. 

Hope this helps.

STAIRB, 2018-04-17:
It helps Thank you very much !

codelec, 2018-04-17:
I am trying to get support for zynq-7000 you can find the initial support files here 
https://github.com/codelec/freedom/tree/pynq 
I am using UDP over ethernet as the debug transport module with the arm core reading the packets and talking to rocket via m_axi_gp0 . For now the max of what I have been able to achieve is reading and writing debug module registers . Executing an ELF on successfully on rocket is WIP

shirlynan, 2019-11-15:
> Hi @STAIRB , Including WithJtagDTM in your core will give you JTAG signals in your ExampleRocketSystem module. they will be called debug_systemjtag_jtag_* ( * = TCK, TMS,TDI,TDO_data, TDO_driven, [...]

Hi, i'm really interested with your jtag implementation with rocket chip. I also tried to debug the core with jtag interface, and connected the generated jtag signals to an external debugger, however it didn't work. Hope u still can read my message and i will be great if u can give me some tipps...

CLappin, 2019-11-19:
Hi,

If you can try your generated core in the Emulator first you should be able to start a debug session there. if you can connect a debug session here you can be certain that the generated hardware is not the issue. 

If you have a look at the readme on the Rocket-Chip github page it will tell you how to set up the emulator. if you look up RBB on that page it should bring  you to the right area. 

sorry  I can be too much more help. 

Thanks, 
Ciaran

michael-etzkorn, 2022-04-12:
Original question has been answered. We can handle FPGA flow in documentation plan. 

https://github.com/chipsalliance/rocket-chip/issues/2960

closed 2022-04-12 by michael-etzkorn