chipsalliance/rocket-chip #1284: vcs build failed

clx7913, 2018-03-14:
I have just already built the C emulator without any problems.Then,I continued to following  the rocket-chip READM.md.After typing "cd ./vsim" and "make ",I got an error like this：

/bin/bash: line 2: vcs: command not found
make: *** [simv-freechips.rocketchip.system-DefaultConfig] Error 127

Does anyone know what's going on? 

Please help me

clx7913, 2018-03-14:
root@vm:/home/chang/rocket-chip/vsim# make
mkdir -p /home/chang/rocket-chip/vsim/generated-src/
cd /home/chang/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/chang/rocket-chip/sbt-launch.jar "runMain freechips.rocketchip.system.Generator /home/chang/rocket-chip/vsim/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfig"
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/chang/rocket-chip/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
Using addons: 
[info] Set current project to rocketchip (in build file:/home/chang/rocket-chip/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running freechips.rocketchip.system.Generator /home/chang/rocket-chip/vsim/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfig
[info] [0.009] Elaborating design...
Interrupt map (2 harts 2 interrupts):
  [1, 2] => dut

/dts-v1/;

/ {
	#address-cells = <1>;
	#size-cells = <1>;
	compatible = "freechips,rocketchip-unknown-dev";
	model = "freechips,rocketchip-unknown";
	L13: cpus {
		#address-cells = <1>;
		#size-cells = <0>;
		L5: cpu@0 {
			clock-frequency = <0>;
			compatible = "sifive,rocket0", "riscv";
			d-cache-block-size = <64>;
			d-cache-sets = <64>;
			d-cache-size = <16384>;
			d-tlb-sets = <1>;
			d-tlb-size = <32>;
			device_type = "cpu";
			i-cache-block-size = <64>;
			i-cache-sets = <64>;
			i-cache-size = <16384>;
			i-tlb-sets = <1>;
			i-tlb-size = <32>;
			mmu-type = "riscv,sv39";
			next-level-cache = <&L7>;
			reg = <0>;
			riscv,isa = "rv64imafdc";
			status = "okay";
			timebase-frequency = <1000000>;
			tlb-split;
			L3: interrupt-controller {
				#interrupt-cells = <1>;
				compatible = "riscv,cpu-intc";
				interrupt-controller;
			};
		};
	};
	L7: memory@80000000 {
		device_type = "memory";
		reg = <0x80000000 0x10000000>;
	};
	L12: soc {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "freechips,rocketchip-unknown-soc", "simple-bus";
		ranges;
		L1: clint@2000000 {
			compatible = "riscv,clint0";
			interrupts-extended = <&L3 3 &L3 7>;
			reg = <0x2000000 0x10000>;
			reg-names = "control";
		};
[... 46 lines cut ...]
	80000000 - 90000000  RWXC memory@80000000

[info] [54.549] Done elaborating.
[success] Total time: 85 s, completed Mar 14, 2018 4:05:59 PM
mkdir -p /home/chang/rocket-chip/vsim/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/chang/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.fir -o /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.v -X verilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.conf -faf /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.anno.json
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
------------------------------------------------------------------------------
Warning: A LegacyAnnotation was automatically converted.
         This functionality will soon be removed. Please migrate to new annotations.
------------------------------------------------------------------------------
Total FIRRTL Compile Time: 157916.2 ms
cd /home/chang/rocket-chip/vsim/generated-src && \
	rm -f /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v && \
	/home/chang/rocket-chip/scripts/vlsi_mem_gen /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.conf >> /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v.tmp && \
	mv /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v.tmp /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v
cd . && \
	rm -rf csrc && \
	vcs -full64 -notice -line +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -timescale=1ns/10ps -quiet +rad +v2k +vcs+lic+wait +vc+list -CC "-I/include" -CC "-I/path/to/install/riscv/toolchain/include" -CC "-std=c++11" -CC "-Wl,-rpath,/path/to/install/riscv/toolchain/lib" /path/to/install/riscv/toolchain/lib/libfesvr.so -sverilog +incdir+/home/chang/rocket-chip/vsim/generated-src +define+CLOCK_PERIOD=1.0 /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.v /home/chang/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v /home/chang/rocket-chip/vsrc/TestDriver.v /home/chang/rocket-chip/vsrc/SimDTM.v /home/chang/rocket-chip/vsrc/SimJTAG.v /home/chang/rocket-chip/vsrc/plusarg_reader.v /home/chang/rocket-chip/vsrc/ClockDivider2.v /home/chang/rocket-chip/vsrc/ClockDivider3.v /home/chang/rocket-chip/vsrc/AsyncResetReg.v  /home/chang/rocket-chip/csrc/SimDTM.cc /home/chang/rocket-chip/csrc/SimJTAG.cc /home/chang/rocket-chip/csrc/remote_bitbang.cc +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN +libext+.v  -o ./simv-freechips.rocketchip.system-DefaultConfig \
	-debug_pp \

/bin/bash: line 2: vcs: command not found
make: *** [simv-freechips.rocketchip.system-DefaultConfig] Error 127

The complete information is above.

seldridge, 2018-03-14:
You're trying to run Synopsys `vcs` but it's not on your path.

clx7913, 2018-03-14:
Whether the Synopsys vcs is included in the rocket-chip repo?

I followed the README.md and I accomplished the building of C emulator.

Then I thought the building of vcs would be like C emulator,so I straightly typed "make" in the vsim folder.

Then I encountered that error.

seldridge, 2018-03-14:
Synopsys VCS is proprietary, closed source software, so no, this is not included. However, for the purposes of just compiling Verilog to a cycle-based emulator binary, Verilator is an open source equivalent.

In other words, for the purposes of functional verification, the emulator and vsim directories are two different approaches to achieve the same goal. Both produce a binary cycle based emulator of your specific rocket chip configuration that you can execute RISC-V binaries on.

clx7913, 2018-03-14:
I didn't realize this kind of reason.

Thank you very much！ ☆(^_^)☆

closed 2018-03-14 by clx7913