chipsalliance/rocket-chip #1285: Running the assembly tests on the C simulator in order to generate waveforms failed

clx7913, 2018-03-15:
Hello,everyone!

I have already build the C simulator that is capable of VCD waveform generation by typing "make debug",and there was no error.
Then I run "make -jN run-asm-tests-debug",the terminal information as below

root@vm:/home/chang/rocket-chip/emulator# make -j1 run-asm-tests-debug
rm -rf output/rv64um-v-mul.vpd.vcd && mkfifo output/rv64um-v-mul.vpd.vcd
vcd2vpd output/rv64um-v-mul.vpd.vcd output/rv64um-v-mul.vpd > /dev/null &
./emulator-freechips.rocketchip.system-DefaultConfig-debug +max-cycles=100000000 +verbose -voutput/rv64um-v-mul.vpd.vcd output/rv64um-v-mul 3>&1 1>&2 2>&3 | /path/to/install/riscv/toolchain/bin/spike-dasm  > output/rv64um-v-mul.out && [ $PIPESTATUS -eq 0 ]
/bin/bash: vcd2vpd: command not found

I am a freshman and don't know what's wrong.
Does this function need other software's supports?

edcote, 2018-03-15:
Make sure you have Synopsys VCS installed and all the required environment variables set.

seldridge, 2018-03-15:
Hmm... this is actually unexpected, though it's apparently been the case for a long, long time. The asm-debug tests _don't need to generate *.vpd and could generate *.vcs_. However, [`RocketTestSuite.scala`](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/system/RocketTestSuite.scala) is doing it this way.

FWIW, @clx7913: you don't _need_ to run these tests. If you want to generate a waveform, you can use the `+verbose` option when running one of these programs.

The README acts like anybody should be able to run `make -J4 run-asm-tests`, but that's dependent on VCS. It fundamentally doesn't need to be, but this may be more convenient to have things configured as such.

This may be worth fixing to generate *.vcs for the emulator makefrags and *.vpd for vsim. Alternatively, the top level README should indicate that this isn't supported for the emulator directory unless you have VCS (it is indicated as such for vsim).

clx7913, 2018-03-16:
I've got it.
Thank you!

closed 2018-03-16 by clx7913

IC-Dream, 2018-06-12:
@clx7913 Have you solved this problem ??? Thanks in advance

HaFred, 2019-11-27:
> FWIW

Hi Seldridge, I thought Verilator in the /emulator repo is not depending on Synopsys VCS, is it?
Would you please suggest a way to generate the waveforms of run-asm-tests w/o VCS?
Thank you very much! @seldridge