chipsalliance/rocket-chip #1289: Keep track for bug fixes?

joisty, 2018-03-16:
Hi,

While I was studying Rocket cores, I come up to following question.
As the Rocket is an open-source project, it is good that various useful patches
are being done daily, and Rocket is becoming more reliable core. However,
as Rocket is not a SW but a HW, at some point it will become a chip and it will
not be able to apply any further patches.
Basically, for most minor updates and patches that just clean-up codes or enhances
performance slightly, it does not really need to be  a big concern. However, from time
to time, there seems to be some patches related to critical functional bugs. 
These are different story.
If you already have your sillicon, of course, you cannot apply the patches even if it
will fix a critical bug. However, you should at least be easy to  keep tracks of which 
bugs will be appearing in your version of sillicon, to apply any kind of SW turn-arounds,
or even at least be aware of. 
This "issues" board possibly could be doing this role, but I do not think this board 
neccessarily holds all and only the "HW bugs".
I think it'd be the best if there already is some kind of bug-tracking platforms (e.g. errata),
that I just don't know yet. Otherwise it'be better if we can start making one.

Thanks.
Jamie

michael-etzkorn, 2022-04-12:
100% agree. WG has started a discussion about organizing tickets.

michael-etzkorn, 2023-01-11:
Closed by #3141 

RC-WG aims to keep a changelog as a resolution to this issue.

closed 2023-01-11 by michael-etzkorn