chipsalliance/rocket-chip #1290: Generate src/main/scala/rocketchip/Configs.scala?

ppoocde, 2018-03-16:
Do I need to manually generate src/main/scala/rocketchip/Configs.scala ?
According to readme, there is a config file to set the parameter of Rocket Chip.
However, I don't see this file under src/main/scala/.
I have src/main/scala/config/Config.scala instead.

CLappin, 2018-03-16:
No, you do not need to make a separate Configs.scala file. The main config file that is used is in the src/main/scala/system/Configs.scala. In this file you will see the default configuration as well as some other example configurations. To build the verilog, then go to the rocket-chip/vsim folder in the terminal and type "make verilog", this will run the default config. To build one your own configs the type "make CONFIG=MyConfig verilog"

Also, there is some more information available from the link below: https://forums.sifive.com/t/getting-started-modifying-the-e300-generating-rtl/1029/3

ppoocde, 2018-03-19:
Thanks Clappin.

If I want to create my own config. say "MyConfig". I need to define Myconfig class inside system/Configs.scale by adding the following line:
class MyConfig extends Config () 

I see Configs.scale already define several class. such as class DefaultConfig and class DualCoreConfig
If I want to generate the core with dual core, then I need to type:
make CONFIG=DualCoreConfig verilog" ?

CLappin, 2018-03-19:
No problem. 
Yup that's how you'd define your own class. And your correct in say it would be 
make CONFIG=DualCoreConfig verilog

ppoocde, 2018-03-20:
Is there any document I can see which item is configurable?
I looked at those configurable item in Configs.scale.
such as DTSModel, ExtMem, they happen to be defined as:
case object ExtMem extends Field
Can I say, for any item define as "case object xxxxx extends Field", they are configurable through Configs.scala?

terpstra, 2018-03-20:
Yes.

closed 2018-03-22 by ppoocde