chipsalliance/rocket-chip #1297: Can L2 cache be removed?

ppoocde, 2018-03-20:
There is a class defined in src/main/scala/subsystem/Configs.scala.
It is: 
class WithNBanksPerMemChannel (n: Int) extends Config
If I don't have to have any L2 cache, should I define 0 to this class?

Also, there is WithNBigCores and WithNSmallCores class define in same file.
What is the different between BigCores and SmallCores? 
I just want a single risc v core without L2 cache in my SOC.

I can't find document describe the parameter of rocket chip. 
Where I can find such document?

Thanks

CLappin, 2018-03-20:
Hi PPoocde,

I don't think that the RocketChip has an L2 cache enabled in it. An older version before moving to TileLink2 did, but I haven't seen one since.  A similar question was asked [here](https://github.com/freechipsproject/rocket-chip/issues/1257)  which may help. 

I believe (and if I'm incorrect please correct me) that the NBigCores allows you  to select how many 64 bit harts are in your "core" where as the NSmallCores is how many 32 bit harts are in it. 

I find using grep -r SEARCHTERM * and finding where each function is defined helps a lot in figuring out what it does. There are some comments in the low level files.  I'm still learning this code base too and it's been extremely helpful!

Thanks, 
Ciaran

ppoocde, 2018-03-21:
Hi Ciaran,

Does Hart mean number of core? 
For example, address and data width I want to be 32bit width and single core for my project, then I should set:
NBigCores to 0
NSmallCores to 1
?

Thanks,
Sunny

CLappin, 2018-03-21:
Hi  Sunny, 

You should look at the DefaultRV32Config  and the withRV32 classes.
Harts are hardware threads, the User Level Spec explains these better. 

thanks,
Ciaran

ppoocde, 2018-03-22:
Thanks again Ciaran. I would have to spend more time on their code and doucment.

closed 2018-03-22 by ppoocde