chipsalliance/rocket-chip #1316: How to implement my own program on bare rocket-chip embedded in the zedboard?

clx7913, 2018-04-03:
I don't have the experience of CPU development before,and just have the experience of simple MCU development,such as MSP430,8051.

I've learned C and assembly language.I just want to implement my own simple program on bare rocket-chip embedded in the zedboard.

My problem is that I don't know the development flow.Firstly I think I need to write down my program by cross compiler,and then how to make the program into the FPGA?

mayoru, 2018-04-10:
For an ASIC, as well for an FPGA, there is a JTAG module in rocketchip. Check sifive projects for the whole assembly. The ARM processor is not needed in case you debug via JTAG.

clx7913, 2018-04-11:
Hello,I have another question.

Could I use the JTAG on zedboard instead of sifive boards?I only have a zedboard now.

mayoru, 2018-04-13:
You can connect to the PL part of your ZYNQ via PMOD or FMC connectors of Zedboard. See Freedom E300 Arty FPGA Dev Kit for the reference. It has the same PMOD connectors as Zedboard.

clx7913, 2018-04-20:
@mayoru   Sorry for my ignorance,I have another question.

Zedboard has two JTAG interfaces:USB-JTAG and 14-pin JTAG.They can't be used to download application ptograms? Must use the PMOD?

mayoru, 2018-04-23:
Those are connected to PL JTAG dedicated pins, inaccessible from programmable logic (see Xilinx UG585,  chapter _JTAG and DAP Subsystem_).

If you want to communicate with rocket-chip via a debugger, you have to connect your JTAG adapter to the debug module of rocket-chip via user I/O pins of Zynq (e.g. connected to PMOD).

If you just want to change the contents of your program memory, you can somehow do it via PL JTAG (either via Cortex or changing FPGA bitstream).

clx7913, 2018-04-23:
Nice,I will check it.

 thanks again.