chipsalliance/rocket-chip #1323: A configuration with a single master AXI bus

mayoru, 2018-04-05:
I used to make a configuration with a single master AXI bus, either AXI4MemPort or AXI4MMIOPort. Now, with a newer version of rocket-chip, a check in Nodes.scala prohibits  of doing so:
Caused by: java.lang.IllegalArgumentException: requirement failed: memory_bus_xbar.node (a nexus) has 1 required inputs and no possible outputs
It seems like either the external mem bus or the scratchpad must be present. Unfortunately, the scratchpad supports nWays = 1 only.

Is there a safe way to remove one of those AXI buses?

ppoocde, 2018-04-06:
I tried to remove slave AXI bus by comment out:
with HasSlaveAXI4Port and,
with HasSlaveAXI4PortModuleTmp
in systemp/ExampleRocketSystem.scala
The DUT in generated verilog does successfully have the ports remove.
I think you can do this for master AXI bus.

mayoru, 2018-04-06:
That works nice for the slave port (actually I do it in my configuration too). Sadly, it doesn't work for HasMasterAXI4MemPort, HasMasterAXI4MMIOPort in case the scratchpad is absent. It worked in some previous version (November 2017), now, with a newer version, I can't do it.
I wonder if WithNMemoryChannels has something to do with it, but setting WithNMemoryChannels(0) doesn't help either.
Thanks for you reply anyways.

hcook, 2018-04-06:
Sorry, I'm a little unclear on what configuration you want, beyond wanting just a single port. Here's a breakdown of combinations:
DScratchpad, MMIO port only: should work with `WithNMemoryChannels(0)`
DScratchpad, Mem port only: might work but is kind of weird and I wouldn't be surprised if its broken, sorry
DCache, MMIO port only: definitely won't work (DCache can't see anything cacheable)
DCache, Mem port only: should work

Which one do you want and which one triggered the requirement failure?

mayoru, 2018-04-09:
Mine is the third combination, so now I see that it shouldn't work.
In fact my primary intention is to have one AXI bus with RAM and IO devices on it.
L1 Cache is to be disabled for IO devices and enabled for RAM (with nWays = 2)
Is it possible to use the fourth combination (nWays for DCache = 2, AXI Mem Bus) and disable cache for a region with IO devices?

ppoocde, 2018-04-09:
Correct me if i am wrong, because my understanding is that anything write to mmio are non-cache transaction. The default config already have two ports, one for memory interface with memory controller, the other port is for MMIO. They should be non-cache.

hcook, 2018-04-10:
@ppoocde Everything attached to the memory port will always appear to be cacheable to the L1. Everything attached to the MMIO port will be non-cacheable MMIO. Cacheability is what differentiates the two ports.

@mayoru Changing rocket-chip to support a new type of port onto which cacheable and non-cacheable regions are both mapped might be possible, but is non-trivial. TileLink insists that there be only a single path through the network per address set, and enforcing that behavior while mapping all addresses down to one port isn't simple. I think you will have an easier time attaching both ports into an AXI crossbar in a wrapper outside of rocket-chip and muxing them down onto one port there.

I will nevertheless make it easier to turn off one port or the other. @mayoru which configuration resulted in the requirement failure?

mayoru, 2018-04-10:
Thanks for your comments, @ppoocde, @hcook.

I see that AXI4MMIOPort alone can't be used with DCache enabled (and this was my case)
I've also checked that both options for one-port configuration are generated properly, that is:
- MMIO port only with DScratchpad
- Mem port only with DCache.

closed 2018-04-10 by mayoru