chipsalliance/rocket-chip #1325: DualCoreConfig: VCS simulator cannot run riscv-tests corrctly.

BenjaminLpn, 2018-04-06:
I tried to build a dual-rocket environment these days. I checked out the master branch of this repository, and the commit id is [8c6e745](https://github.com/freechipsproject/rocket-chip/tree/8c6e745). It was gratifying that verilator simulator run all the riscv-tests and vpd file was dumped successfully. While VCS simulator was also generated, it cannot run benchmarks in riscv-tests, such as towers.riscv. The towers.riscv.out record the execution information.

The command make CONFIG=DualCoreConfig outptu/towers.riscv.vpd is extended to:
```shell
cd . && ./simv-freechips.rocketchip.system-DualCoreConfig -q +ntb_random_seed_automatic +verbose +max-cycles=100000000 output/towers.riscv 3>&1 1>&2 2>&3 | /home/---/riscv/rocket-chip/riscv-tools/riscv-tc/bin/spike-dasm  > output/towers.riscv.out && [ $PIPESTATUS -eq 0 ]
```

Here is log of towers.riscv.out, in which I deleted the duplicate rows.
```c
testing $random d8c356b1 seed 2740741787  
C0:          0 [0] pc=[158ab26a15] W[r 0=0000000000000000][0] R[r 6=0000000000000001] R[r30=8bbddf158bbddf14] inst=[0000f511] bnez    a0, pc - 244  
./simv-freechips.rocketchip.system-DualCoreConfig-debug: invalid option -- 'q'  
C0:         28 [0] pc=[158ab26a15] W[r 0=0000000000000000][0] R[r 6=0000000000000001] R[r30=8bbddf158bbddf14] inst=[0000f511] bnez    a0, pc - 244  
C0:         29 [1] pc=[0000010040] W[r10=0000000000000000][1] R[r 0=0000000000000000] R[r20=8bbddf158bbddf17] inst=[f1402573] csrr    a0, mhartid  
C0:         30 [1] pc=[0000010044] W[r11=0000000000010044][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00000597] auipc   a1, 0x0  
C0:         31 [1] pc=[0000010048] W[r11=0000000000010080][1] R[r11=0000000000010044] R[r28=8bbddf158bbddf17] inst=[03c58593] addi    a1, a1, 60  
C0:         32 [1] pc=[000001004c] W[r 0=000000144a274194][0] R[r 0=0000000000000000] R[r 5=8bbddf158bbddf17] inst=[10500073] wfi (args unknown)  
C0:        110 [0] pc=[0000010050] W[r 0=0000000000010052][0] R[r31=d6abf3afd6abf3af] R[r29=8bbddf158bbddf17] inst=[0000bff5] j       pc - 4  
C0:        111 [1] pc=[0000000800] W[r 0=0000000000000804][1] R[r 0=0000000000000000] R[r12=8bbddf158bbddf17] inst=[00c0006f] j       pc + 0xc  
C0:        113 [1] pc=[000000080c] W[r 0=0000000000000004][0] R[r 0=0000000000000000] R[r31=8bbddf158bbddf17] inst=[0ff0000f] fence  
C0:        114 [1] pc=[0000000810] W[r 0=5186b9a35186b9a3][1] R[r 8=95bef62b95bef62b] R[r18=8bbddf158bbddf17] inst=[7b241073] csrw    dscratch, s0  
C0:        119 [1] pc=[0000000814] W[r 8=0000000000000000][1] R[r 0=0000000000000000] R[r20=8bbddf158bbddf17] inst=[f1402473] csrr    s0, mhartid  
C0:        122 [1] pc=[0000000818] W[r 0=0000000000000100][0] R[r 0=0000000000000000] R[r 8=0000000000000000] inst=[10802023] sw      s0, 256(zero)  
C0:        123 [0] pc=[000000081c] W[r 0=0000000000000400][0] R[r 8=0000000000000000] R[r 0=0000000000000000] inst=[40044403] lbu     s0, 1024(s0)  
C0:        143 [1] pc=[0000000820] W[r 8=0000000000000000][1] R[r 8=0000000000000000] R[r 1=0000000000000003] inst=[00147413] andi    s0, s0, 1  
C0:        144 [1] pc=[0000000824] W[r 0=0000000000000000][0] R[r 8=0000000000000000] R[r 0=0000000000000000] inst=[02041063] bnez    s0, pc + 32  
C0:        148 [1] pc=[0000000828] W[r 8=0000000000000000][1] R[r 0=0000000000000000] R[r20=0000000000000003] inst=[f1402473] csrr    s0, mhartid  
C0:        151 [1] pc=[000000082c] W[r 0=0000000000000400][1] R[r 8=0000000000000000] R[r 0=0000000000000000] inst=[40044403] lbu     s0, 1024(s0)  
C0:        161 [1] pc=[0000000830] W[r 8=0000000000000000][1] R[r 8=0000000000000001] R[r 2=0000000000000003] inst=[00247413] andi    s0, s0, 2  
C0:        162 [1] pc=[0000000834] W[r 0=0000000000000000][0] R[r 8=0000000000000000] R[r 0=0000000000000000] inst=[fc0418e3] bnez    s0, pc - 48  
C0:        166 [1] pc=[0000000838] W[r 0=000000000000083c][1] R[r31=0000000000000003] R[r29=0000000000000003] inst=[fddff06f] j       pc - 0x24  
C0:        168 [1] pc=[0000000814] W[r 8=0000000000000000][1] R[r 0=0000000000000000] R[r20=0000000000000003] inst=[f1402473] csrr    s0, mhartid  
C0:        171 [1] pc=[0000000818] W[r 0=0000000000000100][0] R[r 0=0000000000000000] R[r 8=0000000000000000] inst=[10802023] sw      s0, 256(zero)  
C0:        172 [0] pc=[000000081c] W[r 0=0000000000000400][0] R[r 8=0000000000000000] R[r 0=0000000000000000] inst=[40044403] lbu     s0, 1024(s0)  
C0:        192 [1] pc=[0000000820] W[r 8=0000000000000001][1] R[r 8=0000000000000001] R[r 1=0000000000000003] inst=[00147413] andi    s0, s0, 1  
C0:        193 [1] pc=[0000000824] W[r 0=0000000000000000][0] R[r 8=0000000000000001] R[r 0=0000000000000000] inst=[02041063] bnez    s0, pc + 32  
C0:        194 [0] pc=[0000000844] W[r 0=0000000000000000][0] R[r 2=0000000000000000] R[r 0=0000000000000000] inst=[00000000] addi    s0, sp, 0  
C0:        195 [0] pc=[0000000846] W[r 0=0000000000000000][0] R[r 2=0000000000000000] R[r 0=0000000000000000] inst=[00000000] addi    s0, sp, 0  
C0:        215 [1] pc=[0000000844] W[r 8=95bef62b95bef62b][1] R[r 0=0000000000000000] R[r18=0000000000000003] inst=[7b202473] csrr    s0, dscratch  
C0:        216 [1] pc=[0000000848] W[r 0=0000000000000104][0] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[10002223] sw      zero, 260(zero)  
C0:        217 [1] pc=[000000084c] W[r 0=0000000000000850][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[30000067] jalr    zero, zero, 768  
C0:        238 [1] pc=[0000000300] W[r 0=0000000000000304][1] R[r 0=0000000000000000] R[r24=0000000000000003] inst=[0380006f] j       pc + 0x38  
C0:        240 [0] pc=[0000000338] W[r 0=0000000038804023][0] R[r 0=0000000038804023] R[r 8=95bef62b95bef62b] inst=[38804023] unknown  
C0:        241 [0] pc=[000000033c] W[r 0=0000000000000010][0] R[r 0=0000000000000000] R[r 1=95bef62b95bef62b] inst=[00100073] ebreak  
C0:        261 [1] pc=[0000000808] W[r 0=000000000000080c][1] R[r 0=0000000000000000] R[r20=95bef62b95bef62b] inst=[0340006f] j       pc + 0x34  
C0:        263 [1] pc=[000000083c] W[r 0=000000000000010c][0] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[10002623] sw      zero, 268(zero)  
C0:        264 [0] pc=[0000000840] W[r 0=0000000000000008][0] R[r 2=0000000000000000] R[r 0=0000000000000000] inst=[00000000] addi    s0, sp, 0  
C0:        265 [0] pc=[0000000842] W[r 0=0000000000000008][0] R[r 2=0000000000000000] R[r 0=0000000000000000] inst=[00000000] addi    s0, sp, 0  
C0:        285 [0] pc=[0000000840] W[r 0=0000000000000010][0] R[r 0=0000000000000000] R[r 1=95bef62b95bef62b] inst=[00100073] ebreak  
C0:        290 [1] pc=[0000000800] W[r 0=0000000000000804][1] R[r 0=0000000000000000] R[r12=95bef62b95bef62b] inst=[00c0006f] j       pc + 0xc  
C0:        291 [1] pc=[000000080c] W[r 0=0000000000000004][0] R[r 0=0000000000000000] R[r31=95bef62b95bef62b] inst=[0ff0000f] fence  
C0:        292 [1] pc=[0000000810] W[r 0=95bef62b95bef62b][1] R[r 8=95bef62b95bef62b] R[r18=95bef62b95bef62b] inst=[7b241073] csrw    dscratch, s0  
C0:        299 [0] pc=[0000000814] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r20=95bef62b95bef62b] inst=[f1402473] csrr    s0, mhartid  
C0:        300 [1] pc=[0000000818] W[r 0=0000000000000100][0] R[r 0=0000000000000000] R[r 8=0000000000000000] inst=[10802023] sw      s0, 256(zero)  
C0:        301 [0] pc=[000000081c] W[r 0=0000000000000400][0] R[r 8=0000000000000000] R[r 0=0000000000000000] inst=[40044403] lbu     s0, 1024(s0)  
C0:        321 [1] pc=[0000000820] W[r 8=0000000000000001][1] R[r 8=0000000000000001] R[r 1=0000000000000003] inst=[00147413] andi    s0, s0, 1  
C0:        322 [1] pc=[0000000824] W[r 0=0000000000000000][0] R[r 8=0000000000000001] R[r 0=0000000000000000] inst=[02041063] bnez    s0, pc + 32  
C0:        323 [1] pc=[0000000844] W[r 8=95bef62b95bef62b][1] R[r 0=0000000000000000] R[r18=0000000000000003] inst=[7b202473] csrr    s0, dscratch  
C0:        324 [1] pc=[0000000848] W[r 0=0000000000000104][0] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[10002223] sw      zero, 260(zero)  
C0:        325 [1] pc=[000000084c] W[r 0=0000000000000850][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[30000067] jalr    zero, zero, 768  
C0:        346 [1] pc=[0000000300] W[r 0=0000000000000304][1] R[r 0=0000000000000000] R[r24=95bef62b95bef62b] inst=[0380006f] j       pc + 0x38  
[... cut ...]
  what():  could not open +ntb_random_seed_automatic  
```

If I delete <font background="gray">+ntb_random_seed_automatic</font> in Makefile, the message will be replaced by <font background="gray">+verbose</font>. 
I guess that the `std::runtime_error` is the primary reason which results in stop.

May I ask:
1. Is there anything wrong I used in the command line?
2. Is the error triggered by Debug function?
3. How to fix this error?
4. Is there a stable commit for dual-rocket which can run on FPGA?

Please give me some suggestions. Thanks.

seldridge, 2018-04-06:
> Is there anything wrong I used in the command line?

You're not doing anything wrong here, but that specific commit (and ones before it) have a bug in how `riscv-fesvr` processes command line arguments. Roughly, `riscv-fesvr` previously would error on options that it didn't understand. This was fine for Verilator (as the options are known at the emulator level), but VCS has all sorts of odd options... The solution, discussed here (https://github.com/riscv/riscv-fesvr/pull/44), was to add additional options that would enable/disable strict option parsing via `+permissive-on`/`+permissive-off`. You can see this in newer versions of the vsim makefiles, e.g., https://github.com/freechipsproject/rocket-chip/blob/master/vsim/Makefrag#L83. 

> Is the error triggered by Debug function?

No error with the `debug` function. If you're referring to the `invalid option --q` part, that is fixed in a newer commit.

> How to fix this error?

Bump rocket-chip, rebuild riscv-tools (or at least riscv-fesvr), and everything should work.

> Is there a stable commit for dual-rocket which can run on FPGA?

I can't comment on that, but any of these commits should be fine to produce Verilog that you can map to an FPGA. This is solely a problem with `riscv-fesvr` and no issue with a dual core config.

BenjaminLpn, 2018-04-08:
@seldridge Sorry to bother you that I am confused about other two questions:
1. It seems that both [ucb-bar/fpga-zynq](https://github.com/ucb-bar/fpga-zynq) and [donggyukim/fpga-zynq](https://github.com/donggyukim/fpga-zynq) are suspended, and the newest versions are not compatible with the latest [rocket-chip](https://github.com/freechipsproject/rocket-chip) any more. So may I ask how did you verify the Verilog on FPGA? There may be two feasible ways:
	* Build the FPGA verification platform based on fpga-zynq repository [[1]](https://github.com/ucb-bar/fpga-zynq)/[[2]](https://github.com/donggyukim/fpga-zynq).
	* There is another open-source repository I do not know.

2. Looking at the waveforms of riscv-tests dumped by the VCS simulator (DualCoreConfig), we found that Core 0 executed almost all the work while Core 1 was always waiting for interruption. How to let the two cores execute the process cooperatively? If I want to run Linux with SMP on dual-rocket, is there anything I must config in Chisel or other source files?

Thanks for your time and patience.

seldridge, 2018-04-08:
SiFive provides both Arty and VC707 FPGA shells which are an alternative approach. You could use one of these shells directly or write a new one for your specific FPGA.

Fpga-zynq is essentially deprecated at this point in favor of Midas/Strober/Firesim. These latter approaches should enable running directly on Amazon FPGA instances or on a standalone zedboard.

Not sure about anything specific for getting SMP to do something. I expect that that's 100% software, though. Note: Linux is anecdotally unbootable in simulation (running at ~10KHz). You may just have not gotten to the point where it is doing something useful beyond the boot sequence (this will take hours). Someone else may have some thoughts here...

BenjaminLpn, 2018-04-09:
I found the [repository](https://github.com/sifive/fpga-shells) just now.
Thank you very much.