chipsalliance/rocket-chip #1339: How to debug using the Emulator ?

noureddine-as, 2018-04-12:
Hello,

I built a simple baremetal program [here](https://github.com/noureddine-as/riscv-baremetal-DefaultConfig), (after adding a volatile wait flag and loops); debugging using the interactive Spike is ok. I tried[ OpenOCD and GDB with spike](https://github.com/riscv/riscv-isa-sim#debugging-with-gdb), it also ok (at least they connect). But now I would like to move to a hw simulation under rocket-chip/emulator .. I generated a DefaultConfig emulator and the program works. But I couldn't debug it with the same method as spike (OpenOCD + GDB). I don't what should be changed in this case? Is it possible at the first place?

When launching the emulator it tels me:
```
./emulator-freechips.rocketchip.system-DefaultConfig monocore/gdb_test.101.out  2>&1 | spike-dasm
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 46359
```
I tried to put the ` +jtag_rbb_enable=1` in the parameters but nothing changes where this arg should be put?

Thanks,
Best regards.
Noureddine

noureddine-as, 2018-04-12:
Thanks @mwachs5 I added WithJtagDTMSystem to the DefaultConfig, then make, and the emulator is generated.

Now Trying to debug my program, same as Spike, using OpenOCD + GDB + the Emulator working together. I just have a couple of questions:

1- As you see in the left part of the screenshot 1, a lot of things are printed on the screen rapidly. Should I possibly modify a parameter of the emulator? I mean I want to see only my program executing.

2- The rbb port on the emulator changes each time, i don't see any parameters to modify it? Because since it varies I should each time change the OpenOCD configuration file.

3- The OpenOCD I used is the same as [Spike's](https://github.com/riscv/riscv-isa-sim)  I just change the port according to the one given by the Emulator. But in the second screenshot, I got something like `UNEXPECTED: ...` and some `Error` there .. Is this problematic? Should I modify the .cfg file?

Thanks a lot.
Kind regards.
Noureddine

[image]

[image]

TM1234, 2018-04-12:
How is openocd being run here?
What command line options and/or configuration scripts are being passed to it?
Judging by the openocd log output above whatever script is being used is specifying an IDCODE of 0x10e31913 (the IDCODE for the SiFive E31 RISC-V I think) but whatever is being connected to has an IDCODE of 0x00000001. 
Either the script should be changed to specify 0x00000001 or no IDCODE at all or else the errors should be ignored as innocuous and you continue anyway.
If the errors are ignored then the log output looks fine to me.

mwachs5, 2018-04-12:
To get rid of the `UNEXPECTED` just remove the `expected_id` part of your `.cfg` file. As you can see it pretty much ignores it anyway. This looks like it's working properly and is ready for you to connect via GDB, etc.

To get rid of the "moving around" port (which we do to prevent conflicts when multiple of these are running at a time), you can resolve this `TODO` in your code and put a hard-coded port number:
https://github.com/freechipsproject/rocket-chip/blob/8f3009feb1cc5f0242d17822f4578e1a5f8c3e57/src/main/resources/csrc/SimJTAG.cc#L24

If you want to make a PR to the rocket-chip README.md describing your process, that would be much appreciated!

noureddine-as, 2018-04-13:
@TM1234 I'm using [Spike's config](https://github.com/riscv/riscv-isa-sim#debugging-with-gdb)
Indeed I deleted the IDCODE and this is ok now, thank you.

@mwachs5 I suggest a PR I did [here](https://github.com/noureddine-as/rocket-chip/pull/1), where I didn't change SimJTAG.cc, I just added --rbb-port=PORT parameter in Emulator.cc. That way if the user chooses a particular port he takes it, if not he passes in 0 which makes it choose a random one just as it was initially. I tested the thing and I think it's OK.

Here I suppose that ```DefaultConfigRBB``` is defined as:
```
class DefaultConfigRBB extends Config(new WithNBigCores(1) ++ new BaseConfig ++ new WithJtagDTMSystem)
```

[image]

I still have those annoying messages scrolling around (which doesn't let you see if there is any message printed) (see 1st images in my previous comment).

Concerning the method I'll try to add something about it in the README.md once I get familiarized with the debugging tools.

Thanks a lot all for your help.
Kind regards,
Noureddine

noureddine-as, 2018-04-13:
Hello @mwachs5 @TM1234 !
After a lot of digging and testing i got the following results:

1- Testing[ this app ](https://github.com/noureddine-as/riscv-baremetal-DefaultConfig/tree/gdb-test)in spike is done nicely with spike, openocd and gdb.
2- Testing against the C Emulator doesn't really work :(

  - As you see here is the config file I used with openocd (the same as spike's)
```
interface remote_bitbang
remote_bitbang_host localhost
remote_bitbang_port 9823

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

gdb_report_data_abort enable

init
halt
```

  - GDB and OpenOCD are somewhat slow (I guess it's normal since it's hw emulation, not like spike)
  - For that I've **`  set remotetimeout  200 `** to resolve the following warning from OpenOCD:
```
Info :  hart 0: XLEN=64, 1 triggers
Info : Listening on port 3333 for gdb connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Warn : negative acknowledgment, but no packet pending
Warn : negative acknowledgment, but no packet pending
Warn : negative acknowledgment, but no packet pending
Warn : negative acknowledgment, but no packet pending
...
```
  - Now I still get these even with the set timeout:

 ```
Warn : keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (1460). Workaround: increase "set remotetimeout" in GDB
Warn : keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (1956). Workaround: increase "set remotetimeout" in GDB
...
```

  - These warnings can be neglected I guess **but the main problem is that I don't get the correct results, as shown here  https://github.com/riscv/riscv-isa-sim#debugging-with-gdb  :**

```
(gdb) set remotetimeout 200
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x0000000000010050 in ?? ()
(gdb) set remotetimeout 2000
(gdb) print text
$1 = "`\225\347\271L\347\343Y\001^{\024\234W\v\021%x\201\256\313\310\177\204~6\022f'\372\260Z.\235"
(gdb) print text
$2 = "`\225\347\271L\347\343Y\001^{\024\234W\v\021%x\201\256\313\310\177\204~6\022f'\372\260Z.\235"
(gdb) print wait
$3 = 1046706486

```

How can I resolve this problem?
Is it a problem in the HW implementation of the JTAG RBB or something?
Or may be we need some other configurations there like the [SiFive's one](https://github.com/sifive/freedom-e-sdk/blob/master/bsp/env/freedom-e300-hifive1/openocd.cfg), I tried something similar byt still get the same results.

Best regards,
Noureddine

TM1234, 2018-04-13:
I can't see where you're loading the program to the target?

```
(gdb) set remotetimeout 200
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x0000000000010050 in ?? ()
(gdb) set remotetimeout 2000
(gdb) print text
$1 = "`\225\347\271L\347\343Y\001^{\024\234W\v\021%x\201\256\313\310\177\204~6\022f'\372\260Z.\235"
(gdb) print text
$2 = "`\225\347\271L\347\343Y\001^{\024\234W\v\021%x\201\256\313\310\177\204~6\022f'\372\260Z.\235"
(gdb) print wait
$3 = 1046706486
```

noureddine-as, 2018-04-13:
Oops excuse me @TM1234
(Spike did need that so i completely forgot to do that in this case)

```
(gdb) set remotetimeout 200
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x0000000000010050 in ?? ()
(gdb) load
Loading section .text.init, size 0x2d4 lma 0x80000000
Loading section .tohost, size 0x48 lma 0x80001000
Loading section .text, size 0x9fc lma 0x80001048
Loading section .rodata, size 0x158 lma 0x80001a44
Loading section .rodata.str1.8, size 0x30 lma 0x80001ba0
Loading section .data, size 0x22 lma 0x80001bd0
Loading section .sdata, size 0x4 lma 0x80001bf4
Start address 0x80000000, load size 3782
Transfer rate: 40 bytes/sec, 540 bytes/write.
(gdb) print text
$1 = "Vafgehpgvba frgf jnag gb or serr!"
(gdb) print wait
$2 = 1
(gdb) print wait=0
$3 = 0
(gdb) c
Continuing.
keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (1428). Workaround: increase "set remotetimeout" in GDB
keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (3080). Workaround: increase "set remotetimeout" in GDB
keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (1284). Workaround: increase "set remotetimeout" in GDB
keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (1108). Workaround: increase "set remotetimeout" in GDB
^C
Program received signal SIGINT, Interrupt.
0x00000000800010d8 in main (argc=0, argv=<optimized out>)
    at src/main.c:26
26	    while (!wait)
(gdb) print text
$4 = "Instruction sets want to be free!"
(gdb) 
```
That's it finally. Thank you very much @TM1234  and @mwachs5 .

Best regards,
Noureddine

TM1234, 2018-04-13:
Great - glad it helped.
Megan may know more about the keep_alive() messages but it may be due to slowness of the verilator target compared to real hardware?
In any case it may be possible to just ignore those messages.

closed 2018-06-02 by noureddine-as

MeibaoXu, 2018-06-22:
Hi @TM1234 or @noureddine-as :

I think I follow exactlly with your steps in running the mulator:

1. the emulator is make with CONFIG=DefaultConfigRBB;
2. openocd's '.cfg' file is same as you have posted above;
3. the '.elf' is generated from [riscv-baremetal-DefaultConfig](https://github.com/noureddine-as/riscv-baremetal-DefaultConfig), and ELF is "baremetal-example.out";
4. run the GDB "riscv64-unknown-elf-gdb baremetal-example.out", and set remotetimeout 200, successfully connect to the emulator;

However, I still cannot get the same results running the emulator, the main different is that the 'print text' cannot get the string, but a strange number instead. What I get from the GDB is:

> (gdb) set remotetimeout 200
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x0000000000010050 in ?? ()
(gdb) load
Loading section .text.init, size 0x2d4 lma 0x80000000
Loading section .tohost, size 0x48 lma 0x80001000
Loading section .text, size 0x1778 lma 0x80001048
Loading section .text.startup, size 0x88 lma 0x800027c0
Loading section .rodata, size 0x2b0 lma 0x80002848
Loading section .rodata.str1.8, size 0x30 lma 0x80002af8
Loading section .data, size 0x22 lma 0x80002b28
Loading section .sdata, size 0x4 lma 0x80002b4c
Start address 0x80000000, load size 7714
Transfer rate: 57 bytes/sec, 964 bytes/write.
(gdb) p text
$1 = 1734762838
(gdb) p wait
$2 = 1
(gdb) p wait=0
$3 = 0
(gdb) c
Continuing.
keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (1946). Workaround: increase "set remotetimeout" in GDB
keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (1028). Workaround: increase "set remotetimeout" in GDB
^C
Program received signal SIGINT, Interrupt.
0x0000000080002838 in main ()
(gdb) p text  
$4 = 1953721929
(gdb)

There's a ".text.startup" in the loading section, which you did have, and when interupt the program from processing, there is no information showing which line of the code is beening processing. 
I thought the symbles is not properly loaded, so I tried:

> (gdb) file baremetal-example.out
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from baremetal-example.out...(no debugging symbols found)...done.
(gdb) p text
$6 = 1953721929
(gdb)

Nothing changes.
It is so weird and I cannot figure out what is wrong, can you help with this?

Best Regards.
Meibao

noureddine-as, 2018-06-22:
Hello @MeibaoXu 
From that message
``....`(no debugging symbols found) ...
```
I conclude that you should replace the optimization `-O3` flag with something like `-g -Og -ggdb` or so, this way GDB will recognize all symbols in the program.

Hope this will help.
Best regards,

noureddine-as, 2018-06-27:
Glad it helped @MeibaoXu :)

angi10, 2019-11-05:
@noureddine-as can you help me? 
I followed all the steps of these other users.
When i run GDB and load the file, it appears to me
`Load failed`