chipsalliance/rocket-chip #1343: How can I use performance counters in RocketCore.scala code?

sj-leo, 2018-04-13:
In [rocket-chip/src/main/scala/rocket/RocketCore.scala](https://github.com/freechipsproject/rocket-chip/blob/93bf8e62b58e39dbe838488d1fecd4aea4d3102d/src/main/scala/rocket/RocketCore.scala#L61) [line 64-106], there are a lot of performance counters to profile the events such as load, store, branch, arithematic operations, FLOPs, cache miss and TLB miss.

However, I don't find detailed document how I use these counters.
So, I would be grateful if you could tell me how to use it in detail or if you can give me some reference about that.

solomatnikov, 2018-04-13:
https://static.dev.sifive.com/U54-MC-RVCoreIP.pdf, section 3.11

sj-leo, 2018-04-13:
@solomatnikov First of all, thank you for your reply.

We try to use the performance counters with your reference, but we have many difficulties. Please answer the following questions.

As far as I know, we should use the M (Machine)-mode in order to use a hardware performance monitor. However, I don't know where we can change the mode and where we use the command ‘csrr’ to read mhpmcounter.

So, if you shared the sample code, we will be very helpful in solving the problem.
Thank you.

sequencer, 2022-04-08:
need documentation and test on performance counter. Closing for know.

closed 2022-04-08 by sequencer

sammy17, 2022-08-04:
> 

I tried this configuration in Rocket core DefaultConfig by using the below code. But counters never changed. Seemed like counters were initialized with a random value for each simulation. 

Please let me know if there are any other instructions on how to enable performance counters in Rocket core.

michael-etzkorn, 2022-08-04:
Hi @sammy17, your post will have more visibility on StackOverflow or in our github discussion, and we can move a conversation there.

sammy17, 2022-08-05:
Hi @michael-etzkorn , Thanks for the response. I asked the same in StackOverflow but no response so far. Let me try github discussions.