chipsalliance/rocket-chip #1346: setResource(/vsrc/[..]) side effect

edcote, 2018-04-14:
@grebe Small heads up, this update causes weird behavior for me:

https://github.com/freechipsproject/rocket-chip/blob/8f3009feb1cc5f0242d17822f4578e1a5f8c3e57/src/main/scala/devices/debug/Periphery.scala#L113

The above file (and others) are now copied to my chisel run directory (targetDirName) when emitOneFilePerModule is set. 

I see they are identified in firrtl_black_box_resource_files.f, but there is no corresponding firrtl_design_files.f

Previously my "design_files.f" consisted of *.v in the output directory, manually excluding TestHarness.

I would need to update my algo populate my "design_files.f" to do *.v, but manually exclude TestHarness + all files in firrtl_black_box_resource_files.f

Is this not causing issues with building the emulator on your end? That is the symptom here.  I am not building with your Makefile.

grebe, 2018-04-15:
firrtl_design_files.f is probably something that should change- imho, it should probably be a .d for Makefiles to consume. @jackkoenig, was this something you were working with?

The Makefile included with rocketchip manually specifies every file included in the build. At the moment, it doesn't actually use the SimDTM.v in the targetDirName, it still uses src/main/resources/vsrc/. I had a PR that parsed firrtl_black_box_resource_files.f and included the blackboxes, but it was too big so I've been breaking it up into chunks.

Are you sure you don't want any of the black boxes in your build? How are you pulling them in otherwise?

edcote, 2018-04-16:
Is there a firrtl_design_files.f already?  I didn't see one.

I wouldn't make any assumptions that users are using the Makefile to build rocket, nor they are using the targetDirName that the built-in generator uses.

I disagree with it should be a .d file for Makefile inclusion.  The defacto standard for verilog lists are .f or .flist files.  Result of quick Google search for [info](https://stackoverflow.com/a/25440355/3736700).  You can always output both.

Yes, absolutely, I have the black boxes in my build.  But I maintain different .f files depending on the type of verilog file.

hcook, 2018-04-17:
I too had to update my "algorithm" to populate my "design_files.f". This involved adding a `ls ... -I '*.f'` flag somewhere in my build flow, so wasn't especially onerous.

It looks like you can set a FIRRTL annotation (`BlackBoxTargetDirAnno`) to put the blackboxes in a separate directory from the other build files, but you'll still get a `.f` file in that directory...

I'll open a firrtl issue to discuss how best to resolve this.

grebe, 2018-04-18:
Ah, I think I didn't entirely understand what you were writing about.

@edcote fair enough about Makefiles and .d vs. .f. I think it would be good to have some sort of standard way of translating .f -> makefile dependencies, but I think that is outside the scope of this issue.

To be clear, are you saying you want to include the firrtl_black_box_resource_files.f files in your build, or exclude it like the TestHarness? I'm not sure I'm parsing correctly.