chipsalliance/rocket-chip #1352: doubts about L1 cache configuration

clx7913, 2018-04-16:
For example,my L1 cache has the configuration:"NSets=64,NWays=4".How many bytes or bits each way has?Should I refer to the RowBits or the CacheBlockBytes in /coreplex/Configs.scala?

hcook, 2018-04-16:
`CacheBlockBytes` controls the size of each cache block (of which there are ways*sets).

`RowBits` controls the dimensions of the data SRAM.

clx7913, 2018-04-16:
You mean each way has `CacheBlockBytes` bytes?

If the default value of `CacheBlockBytes` is 64,so the size of the L1 cache is `sets*ways*CacheBlockBytes=64*4*64`.Am I right?

hcook, 2018-04-17:
It depends what you mean by "each way". Each way of a set is `CacheBlockBytes` of data. Each way of a cache is therefore `CacheBlockBytes*sets`. The default cache size is 16Kb.

clx7913, 2018-04-18:
Thank you so much!That's really detailed.

closed 2018-04-18 by clx7913

cabingh, 2018-09-08:
Hello I am still learning the source code structures. where I can find this default CacheBlockBytes setting? I don't see  /coreplex/Configs.scala in current src/main/scala. Thanks in advance.

aignacio, 2018-11-15:
@cabingh I'm not sure about that but [check this line](https://github.com/freechipsproject/rocket-chip/blob/05ccbffaa926cc157665b02c7e0d56f4c4186e14/src/main/scala/subsystem/MemoryBus.scala#L13) where it's defined as 64 bytes. As an example, the configuration of **L1** for **TinyCore** (If I'm right) it's:
D$ cache = 256 **nSets** * 1 **nWays** * 64 **CacheBlockbytes** = 16384 Bytes = **16 kiB**
I$ cache = 64 **nSets** * 1 **nWays** * 64 **CacheBlockbytes** = 4096 Bytes = **4 kiB**
...using the image below (different cfg) we can say that by default we have block size equals to 512 bits instead of 32-bit like image.
[image]
is this right @aswaterman ?

aswaterman, 2018-11-15:
@aignacio that sounds correct to me.

jerryhethatday, 2020-11-08:
> `CacheBlockBytes` controls the size of each cache block (of which there are ways*sets).
> 
> `RowBits` controls the dimensions of the data SRAM.

can you further explain the rowbits field, I have no idea what's that for? what does dimensions of the data SRAM mean?

aswaterman, 2020-11-08:
How wide the data SRAM physically is.  For example, if RowBits is 128 and CacheBlockBytes is 64 (=512 bits), then each cache line spans 4 rows.

It's essentially a physical-design tuning parameter.