chipsalliance/rocket-chip #1354: Remove debug module

ppoocde, 2018-04-17:
On the generated verilog file, there is debug module debug_1 instantiated @ ExampleRocketSystem.
Does anyone know how to remove this from scala? so that it wont be generated on verilog file.
We are trying to reduce the area of the design.

Thanks,

Sunny

mwachs5, 2018-04-17:
1. Comment out `HasPeripheryDebug` in https://github.com/freechipsproject/rocket-chip/blob/20a88768560ca810c0dbe79f1a33604268d921c7/src/main/scala/subsystem/RocketSubsystem.scala#L36

2. Fix the fact that you no longer have `debug` inside that code block (you'll need to find some way to properly tie the debug interrupt to 0).

3. Fix whatever issues arise (there isn't a supported knob to remove this, since it's pretty intrinsic to how the code is loaded to the core, etc).

Note that you *can* significantly reduce the size of the Debug Module. See https://github.com/freechipsproject/rocket-chip/pull/1201/files. You can probably turn Progbuf words down to 2 if you don't care about passing rocket-chip regressions.

ppoocde, 2018-04-18:
@mwachs5 Thanks for the suggestion.
change the Progbuf from 16 to 2 result around 50% area improvement.

If i  want to remove the entire module, I make the following changes:
1. comment out "with HasPeripheryDebug" in trait HasRocketTiles in RocketSubsystem.scala
2. comment out "with HasPeripheryDebugModuleImp" in trait HasRocketTilesModuleImp in RocketSubsystem.scala
3. comment out "dut.connectDebug(clock, reset, io.success)" in class TestHarness in TestHarness.scala
4. change dut.reset := reset | dut.debug.ndreset to dut.reset := reset in class TestHarness in TestHarness.scala
5. comment out "rocket.intInwardNode := rocket { IntSyncCrossingSink(3) } := debug.intnode

With the above change, the input port ".auto_intsink_in_sync_0" in RocketTile tile, is disappear. 
I assume this input port take in the debug interrupt. 
However, when I look inside the verilog code down to core level, 
io_interrupts_debug of Rocket core is connected. I traced the wire to upper level and 
I found that it is eventually connect to output .auto_int_out_0 of CLINT clint.

Do you know what am I missing?

ppoocde, 2018-04-19:
I further removed the logic: 
1. val d_interrupts @ rocket/CSR.scala
2. remove logic that use io.interrupts.debug @ rocket/CSR.scala

Now I no longer see io_interrupts_debug in core level. 
However, what I dont understand is that there is still two interrupt input in TLPLIC and the output of TLPLIC used to have two interrupts output:
auto_int_out_0 and auto_int_out_1
With my changes above, now the module only have two interrupt input and one interrupt output:
auto_int_out_0

codelec, 2018-04-19:
there is a comment on line 112 in subsystem/RocketSubsystem.scala stating that the order of connections to rocket.intInwardNode is important. So even after commenting out `rocket.intInwardNode := rocket { IntSyncCrossingSink(3) } := debug.intnode` you are infact tieing msip interrupt coming from clint to debug interrupt internal to rocket. To avoid that you need to edit tile/Interrupts.scala and remove async_ips from the line where interrupts sinking into node are connected to interrupt lines of rocket
`(async_ips ++ periph_ips ++ seip ++ core_ips).zip(interrupts).foreach { case(c, i) => c := i }` to 
`(periph_ips ++ seip ++ core_ips).zip(interrupts).foreach { case(c, i) => c := i }` and also tie debug interrupt `core.debug := 0.U`

ppoocde, 2018-04-20:
@codelec  I made the change and the interrupt is properly connected. Thanks for your help.

closed 2018-04-20 by ppoocde