chipsalliance/rocket-chip #1355: Config: WithInCoherentTiles

ppoocde, 2018-04-17:
Hi everyone, 
I found this config on system/Config.scala.
If I don't define this in my config, the generated verilog will have TLBroadcast module in Top level DUT.
If I define WithInCoherentTiles ++ in my config, this module will disappear.
In what situation we need to have this module, and what does it do?

Thanks,

Sunny

hcook, 2018-04-18:
The `TLBroadcast` module is a hub that provides coherence between caching master agents, such as L1DCaches. It also keeps them coherent with respect to accesses made to cacheable memory from e.g. the Front Port. In the current `ExampleRocketSystem` it is safe to remove this module using `WithIncoherentTiles` if you have 1) multiple cores with scratchpads instead of data caches, 2) a single data cache with no front port, or 3) you don't want hardware coherence in the system (I don't recommend this!).  There is also a "bufferless" version of the broadcast hub that uses no additional state elements (`WithBufferlessBroadcastHub`) if you are concerned about area.

hcook, 2018-04-23:
1) Yes, all must be scratchpad for data (partial ICache is ok because it is read-only and can be flushed).
2) Yes, only one *caching* master.

Yes, that sounds safe.

ppoocde, 2018-04-24:
@hcook Thanks for your help.

closed 2018-04-24 by ppoocde

ohadbox, 2019-12-11:
Provided I'm using this config flag in my design and I don't see any changes to the generated Verilog script, what am I doing wrong?
What exactly are the changes expected from just using this configuration without other changes to the design.

Background:
I'm trying to remove coherency from a multi-tile architecture. It is done for academic reasons, and I'm trying to do minimum changes and pass on the responsibility of non-coherence in the system to software developers in my team.

Thanks.